zephyrproject-rtos/zephyr #15668: Pull Request hint documentation warning

chinli, 2019-04-25:
I submit a PR with https://github.com/zephyrproject-rtos/zephyr/pull/15636, but it comment one waring always, detail infor as below:
warning: reached end of file while inside a verbatim block!
The command that should end the block seems to be missing!

some one could help me? i have tried many times, but still not resolved.
tks~

Thea-Aldrich, 2019-04-25:
Hello,
Thanks so much for reaching out and thank you for your contribution! Welcome! I have added David Kinder who helps with our documentation. We'll look through the PR and help find the issue that is hanging everything up. 

I also wanted to mention that the Slack Channel is also a great place to get quick help on issues like this. 

https://join.slack.com/t/zephyrproject/shared_invite/enQtNDgwNzk5MTU3NTA3LWNhZjc4MzM4Nzg5NGRiNTA2OGVhN2ZlNzZmYWE5Yjc2ZDIwYjYzNGJmMzNiYmY2MTdiYmQ1M2I4NGEzMmUzZjA

Best,
Thea

dbkinder, 2019-04-25:
The unterminated verbatim block error here is because ``\r`` is meaningful to doxygen comment processing.  Change this to be ``\\r\\n`` and the error should go away.

chinli, 2019-04-26:
@Thea-Aldrich @dbkinder well, thanks for your comment! it's done.

closed 2019-04-26 by chinli