zephyrproject-rtos/zephyr #15672: net: socket send return error(-110) when http request 100 times

chinli, 2019-04-25:
we implement http client based BSD socket, and we request repeatedly when socket connected for testing the stability, but i'm confused that it send error when request 100times almost each test.
"are you kidding me?"
"No, I'm serious, the test procedures as below:"
1. Got IP addr, and connect it with socket
2. http request 
3. recv the http response and parse it
repeates step 2~3,  send error(-110) for the 100th time.

zephyr version：V1.14.0-206
board：96board IVY5661

tbursztyka, 2019-04-25:
That does not tell much, no logs, no code... It could be your test that does something wrong as well.

Could you share the code?

jukkar, 2019-04-25:
The error code 110 is ESHUTDOWN in newlib (you are using that right?).

chinli, 2019-04-26:
@tbursztyka the code of http client you can reference PR #15636, and  I enable all the LOG of networking, but there was any useful information, but i can see it shutdown connection by itself with wireshark.

@jukkar i don't know what do you mean using that right about the newlib. Is there any special usage about socket?

chinli, 2019-04-26:
@tbursztyka there is some logs indicated socket disconnect..
[00:00:28.370,000] <dbg> net_http_client.http_client_request: (0x0010699c): http client request start

[00:00:28.370,000] <dbg> net_if.net_if_tx: (0x0010b5f8): Processing (pkt 0x0011ae20, prio 0) network packet
[00:00:28.370,000] <dbg> net_if.net_if_tx: (0x0010b5f8): Calling context send cb 0x0010ab18 status 178
[00:00:28.380,000] <dbg> net_conn.net_conn_input: (0x0010b560): Check TCP listener for pkt 0x0011a9a0 src port 80 dst port 49759 family 1
[00:00:28.380,000] <dbg> net_conn.net_conn_input: (0x0010b560): [1] match found cb 0x02033a05 ud 0x0010ab18 rank 0x33
[00:00:28.380,000] <dbg> net_http_client.http_client_request: (0x0010699c): http client recv response, recved=363

[00:00:28.380,000] <dbg> net_http_client.on_status: (0x0010699c): HTTP response status Partial Content
[00:00:28.380,000] <dbg> net_http_client.on_header_value: (0x0010699c): http response content_len=64

[00:00:28.380,000] <dbg> net_http_client.on_body: (0x0010699c): on_body: Processed 64 length 64
[00:00:28.380,000] <dbg> net_http_client.http_client_request: (0x0010699c): http recv response complete

[00:00:28.380,000] <dbg> net_if.net_if_tx: (0x0010b5f8): Processing (pkt 0x0011ae20, prio 0) network packet
[00:00:28.380,000] <dbg> net_if.net_if_tx: (0x0010b5f8): Calling context send cb 0x0010ab18 status 54
[00:00:28.380,000] <dbg> net_conn.net_conn_input: (0x0010b560): Check TCP listener for pkt 0x0011a9a0 src port 80 dst port 49759 family 1
[00:00:28.380,000] <dbg> net_conn.net_conn_input: (0x0010b560): [1] match found cb 0x02033a05 ud 0x0010ab18 rank 0x33
[00:00:28.380,000] <dbg> net_if.net_if_tx: (0x0010b5f8): Processing (pkt 0x0011ae20, prio 0) network packet
[00:00:28.390,000] <dbg> net_if.net_if_tx: (0x0010b5f8): Calling context send cb 0x0010ab18 status 54
[00:00:28.390,000] <dbg> net_conn.net_conn_input: (0x0010b560): Check TCP listener for pkt 0x0011a9a0 src port 80 dst port 49759 family 1
[00:00:28.390,000] <dbg> net_conn.net_conn_input: (0x0010b560): [1] match found cb 0x02033a05 ud 0x0010ab18 rank 0x33
[00:00:28.390,000] <dbg> net_conn.net_conn_unregister: (0x0010b560): [1] connection handler 0x0010b8bc removed

jukkar, 2019-04-26:
> i don't know what do you mean using that right about the newlib. Is there any special usage about socket?

As the error code is ESHUTDOWN, it is returned by tcp code when the connection is about to be terminated but user data is tried to be sent.

chinli, 2019-04-26:
@jukkar Yes, I understand  this point. but the question is i don't know what problem break off the socket connection.  there is a log for notice connection be removed as below: 
**[00:00:28.390,000] net_conn.net_conn_unregister: (0x0010b560): [1] connection handler 0x0010b8bc removed**
do you have any ideas?

jukkar, 2019-04-26:
That unregister call is probably happening when we receive SYN and start to establish a connection. See `net_tcp_unregister()` call in tcp.c around line 2149. There might be better information shown if you enable TCP debugging.

chinli, 2019-04-26:
[00:01:10.760,000] <dbg> net_core.processing_data: (0x0010b0f4): Consumed pkt 0x0011cd68
[00:01:10.760,000] <dbg> net_core.net_recv_data: (0x0010711c): prio 0 iface 0x001207c0 pkt 0x0011cd68 len 694
[00:01:10.760,000] <dbg> net_core.net_rx: (0x0010b0f4): Received pkt 0x0011cd68 len 694
[00:01:10.760,000] <dbg> net_conn.net_conn_input: (0x0010b0f4): Check TCP listener for pkt 0x0011cd68 src port 80 dst port 63340 family 1
[00:01:10.760,000] <dbg> net_conn.net_conn_input: (0x0010b0f4): [1] match found cb 0x020332f5 ud 0x0010a6ac rank 0x33
[00:01:10.760,000] <dbg> net_tcp._tcp_established: (0x0010b0f4): DATA received from 192.168.1.102 port 80
[00:01:10.760,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4): [0x0010b4e8] pkt 0x0011d1a0 src 63340 dst 80
[00:01:10.760,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4):   seq 0xbb5392a2 (3142816418) ack 0xc51e5f46 (3307102022/640)
[00:01:10.760,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4):   flags uAprsf
[00:01:10.760,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4):   win 640 chk 0xbf17
[00:01:10.760,000] <dbg> net_tcp.print_send_info: (0x0010b0f4): ACK sent to <log_strdup alloc failed> port 80
[00:01:10.760,000] <dbg> net_http_client.http_client_request: (0x0010659c): http client recv response, recved=640

[00:01:10.760,000] <dbg> net_http_client.on_body: (0x0010659c): on_body: Processed 975 length 640
[00:01:10.770,000] <dbg> net_http_client.http_client_request: (0x0010659c): http recv response continue

[00:01:10.770,000] <dbg> net_core.processing_data: (0x0010b0f4): Consumed pkt 0x0011cd68
[00:01:10.770,000] <dbg> net_core.net_recv_data: (0x0010711c): prio 0 iface 0x001207c0 pkt 0x0011cd68 len 103
[00:01:10.770,000] <dbg> net_core.net_rx: (0x0010b0f4): Received pkt 0x0011cd68 len 103
[00:01:10.770,000] <dbg> net_conn.net_conn_input: (0x0010b0f4): Check TCP listener for pkt 0x0011cd68 src port 80 dst port 63340 family 1
[00:01:10.770,000] <dbg> net_conn.net_conn_input: (0x0010b0f4): [1] match found cb 0x020332f5 ud 0x0010a6ac rank 0x33
[00:01:10.770,000] <dbg> net_tcp._tcp_established: (0x0010b0f4): DATA received from 192.168.1.102 port 80
[00:01:10.770,000] <dbg> net_tcp.net_tcp_change_state: (0x0010b0f4): [0x0010b4e8] state ESTABLISHED (4) => CLOSE_WAIT (5)
[00:01:10.770,000] <dbg> net_tcp.net_tcp_change_state: (0x0010b0f4): [0x0010b4e8] state CLOSE_WAIT (5) => LAST_ACK (6)
[00:01:10.770,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4): [0x0010b4e8] pkt 0x0011d1a0 src 63340 dst 80
[00:01:10.770,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4):   seq 0xbb5392a2 (3142816418) ack 0xc51e5f78 (3307102072/50)
[00:01:10.770,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4):   flags uAprsF
[00:01:10.770,000] <dbg> net_tcp.net_tcp_trace: (0x0010b0f4):   win 1231 chk 0xbc95
[00:01:10.770,000] <dbg> net_tcp.print_send_info: (0x0010b0f4): ACK sent to 192.168.1.102 port 80
[00:01:10.770,000] <dbg> net_http_client.http_client_request: (0x0010659c): http client recv response, recved=49

[00:01:10.770,000] <dbg> net_http_client.on_body: (0x0010659c): on_body: Processed 1024 length 49
[00:01:10.780,000] <dbg> net_http_client.http_client_request: (0x0010659c): http recv response complete

[00:01:10.780,000] <dbg> net_core.processing_data: (0x0010b0f4): Consumed pkt 0x0011cd68
[00:01:10.780,000] <dbg> net_core.net_recv_data: (0x0010711c): prio 0 iface 0x001207c0 pkt 0x0011cd68 len 60
[00:01:10.780,000] <dbg> net_core.net_rx: (0x0010b0f4): Received pkt 0x0011cd68 len 60
[00:01:10.780,000] <dbg> net_conn.net_conn_input: (0x0010b0f4): Check TCP listener for pkt 0x0011cd68 src port 80 dst port 63340 family 1
[00:01:10.780,000] <dbg> net_conn.net_conn_input: (0x0010b0f4): [1] match found cb 0x020332f5 ud 0x0010a6ac rank 0x33
[00:01:10.780,000] <dbg> net_tcp._tcp_established: (0x0010b0f4): DATA received from 192.168.1.102 port 80
[00:01:10.780,000] <dbg> net_tcp.net_tcp_change_state: (0x0010b0f4): [0x0010b4e8] state LAST_ACK (6) => CLOSED (0)
[00:01:10.780,000] <dbg> net_conn.net_conn_unregister: (0x0010b0f4): [1] connection handler 0x0010b450 removed
[00:01:10.780,000] <dbg> net_core.processing_data: (0x0010b0f4): Consumed pkt 0x0011cd68
[00:01:10.850,000] <dbg> net_http_client.http_client_get: (0x0010659c): http client send request:GET /ota/zephyr-signed-ota-96b_ivy5661.bin HTTP/1.1
HOST: 192.168.1.102
Connection: Keep-alive

chinli, 2019-04-28:
Well, tks for your comments, I found the reason of this problem that http protocal Keep-Alive limited 100 times with current connection of http server. so this issue is a normal disconnect.

closed 2019-04-28 by chinli