zephyrproject-rtos/zephyr #15676: Support instrumentation for time spent in various power states

pabigot, 2019-04-25:
A basic need in low-power devices is to compare time spent with the CPU active with time spent with the CPU off.

I can't see anything in Zephyr that provides this capability.  There are subsystems to explicitly control system or device power, but nothing that tracks how much time the idle task is active.

Instrumenting the idle task is trivial but produces implausible results: the system is active for 50% of the first two seconds in an application that does nothing but printk duty cycle information at 1 Hz, then goes on to run for over four minutes without accumulating a single 30.5 us cycle of active time.  So it's clearly not measuring what's desired.

There are also `__WFE` loops sprinkled throughout the driver and bluetooth subsystems that turn off the CPU and bypass this sort of instrumentation.  Other architectures may involve similar tricks.

It would really be nice if this information was available.  If it could be tracked on both the system and device-specific power management that would also be useful.

mbolivar, 2019-05-12:
How practical is this to implement? We may imagine low-power systems where entering deep sleep involves turning off the very clocks we would need to do this tracking.

pabigot, 2019-05-12:
@mbolivar We can also imagine that impossible situations can be declared out of scope.  That doesn't eliminate the value of doing what can be done.

I hadn't intended to clutter up this issue with references to unmergeable patches in my fork, but you can see how trivial it is to get a basic estimate of idle versus active time.  It already reveals behavior that isn't otherwise visible: to wit, why does a trivial application spend 500 ms in active mode on startup?

In any case, if nobody else would find value in instrumenting Zephyr that's fine.

mbolivar, 2019-05-13:
@pabigot I'm not saying "it's not possible"; I'm asking "is it possible?".

My question was an honest one. Have you looked at a variety of different SoCs and concluded that such an API is implementable and covers enough interesting use cases to be introduced as a general purpose API? Your "unmergeable" patches referenced here don't answer that question for me. Sorry if this is such an obvious question that you don't consider it worth answering.

I could also imaging adding trace points for PM transitions and relying on an always-on clock in the tracing environment to capture this information in a generic way that would definitely work on all traceable hardware, at the cost of requiring trace infrastructure. So there's a tradeoff space here and I'm just trying to understand how much research you've done before submitting the issue.

THanks.

pabigot, 2019-05-13:
@mbolivar Sorry, I misinterpreted your response.

No, I have not investigated a variety of SoCs.  I do this in [nrfcxx](https://pabigot.github.io/nrfcxx/classnrfcxx_1_1systemState.html#a3d1956a471409a35f2216b7d0836ff40) where code that changes modes related to active/sleep, high-frequency clock on/off, and radio activity records time in each mode, which is reported in telemetry beacons.

I wasn't intending to propose any sort of solution, just pointing out what I see as a functional gap that's of significant interest when verifying low power solutions.  Generally Zephyr seems to lack support for tracking things like active/idle time, packets transmitted/received, flash reads/writes, etc., which are often helpful in monitoring installations and identifying problems.

nashif, 2021-09-25:
this should be possible now with tracing.

closed 2021-09-25 by nashif