zephyrproject-rtos/zephyr #15679: Can GPTP support multiple slave nodes

Jxudong, 2019-04-26:
When I use one master node and one slave node is working well, when I use one master node and  multiple slave nodes, there is a problem, the slave node report warning is as follows：

[image]

Is this a configuration problem or does GPTP not support multiple child nodes?

jukkar, 2019-04-26:
Are you using TSN supported Ethernet switch? Note that child nodes should not see each other traffic, only master should see child nodes traffic. This means that if you connect three TSN capable devices into normal Ethernet switch, there will be conflicts and it will not work.

Jxudong, 2019-04-28:
This means that if I want to run GPTP, I have to use a switch with TSN functionality, not a normal switch?

Jxudong, 2019-04-28:
@jukkar

jukkar, 2019-05-01:
TSN capable switch is only needed if you have more than two nodes that you want to connect together.
You can overcome this by using VLANs which can be used to separate Ethernet traffic. Using VLANs with gPTP is not specified in gPTP spec but both Linux and Zephyr support that.

ioannisg, 2019-07-30:
@Jxudong please, re-open, if needed , to continue discussion

closed 2019-07-30 by ioannisg