zephyrproject-rtos/zephyr #15696: [question] why bt_setting is dependant of printk in menuconfig?

summer20100514, 2019-04-26:
It's kind of weird that bt setttings is dependant of printk as follows
```
Help:

  When selected, the Bluetooth stack will take care of storing
  (and restoring) the Bluetooth state (e.g. pairing keys) and
  configuration persistently in flash.

  When this option has been enabled, it's important that the
  application makes a call to settings_load() after having done
  all necessary initialization (e.g. calling bt_enable). The
  reason settings_load() is handled externally to the stack, is
  that there may be other subsystems using the settings API, in
  which case it's more efficient to load all settings in one go,
  instead of each subsystem doing it independently.

Direct dependencies (=y):
     SETTINGS(=y)
  && PRINTK(=y)
  && BT_HCI_HOST(=y)
  && BT_HCI(=y)
  && BT(=y)
```

nvlsianpu, 2019-04-26:
cc @jhedberg

nvlsianpu, 2019-04-26:
AFIK sprintk() is used for generating settings key-names form BLT addresses etc (for instance).

jhedberg, 2019-04-26:
Yep, `snprintk` is being used, and there's currently no separation of the printk console output functionality from its formatting functionality, but just one high-level `PRINTK` Kconfig option.

summer20100514, 2019-04-28:
OK, get it.

closed 2019-04-28 by summer20100514