zephyrproject-rtos/zephyr #15698: bluetooth: bt_conn: No proper ID handling

MarekPieta, 2019-04-26:
**Describe the bug**
bt_conn->id is probably not handled properly (using hardcoded value - zero) - noticed errors connected with inproper BLE connection handling while using directed advertising (e.g. unable to properly refer to the connection).

**Expected behavior**
ID should be updated correctly for BLE connections (it should not have hardcoded value of zero).

**Impact**
 It may cause errors in applications - e.g. problems with direct advertising.

jhedberg, 2019-04-26:
Wasn't this fixed by #15532 ?

jhedberg, 2019-04-26:
@pdunaj @carlescufi FYI

Another note: non-default id is only supported for peripheral role, so there should be no need to add custom id functionality for central-related code.

MarekPieta, 2019-04-26:
We didnt set conn->id and it caused problems with direct advertising. Connection was not properly handled by API, which caused problems like advertising failing to start.

jhedberg, 2019-04-26:
@MarekPieta thanks for clarifying, your updated PR is looking better (and simpler :)

pdunaj, 2019-04-29:
Hi @jhedberg , the previous one caused resource leak but I admit I have not tested it enough and there was another bug there.

PR #15700 (2019-04-29): bluetooth: Fix bt_conn->id handling
Change fixes conn->id handling in le connections. 

Fixes #15698
commit 28689c3b (2019-04-29): bluetooth: Fix bt_conn->id handling

Fixed bt_conn->id handling in bt_conn_create_slave_le function.

Fixes: #15698
commit 265c741a (2019-04-29): bluetooth: Fix bt_conn->id handling

Fixed bt_conn->id handling in bt_conn_create_slave_le function.

Fixes: #15698
commit 0654c332 (2019-04-29): bluetooth: Fix bt_conn->id handling

Fixed bt_conn->id handling in bt_conn_create_slave_le function.

Fixes: #15698
closed 2019-04-29 by jhedberg