zephyrproject-rtos/zephyr #15703: ADC (and SPI?) async operations are not really async

pabigot, 2019-04-26:
The `adc_context.h` common file used to support asynchronous ADC requests in a driver-independent manner is only asynchronous for the first operation, as the lock used to queue a new operation is held by the driver as long as the previous operation is active.  This makes the invoking thread block while attempting to schedule the next operation.

The following example may clarify.  This code on nRF52840 queues an ADC sample that takes 72 ms to complete (256x oversampling, ADC init not provided).

```
#define NUM_ASYNC 4

struct k_poll_signal adc_sig[NUM_ASYNC];
struct k_poll_event adc_evt[NUM_ASYNC];
struct adc_sequence adc_seq[NUM_ASYNC];

// Initialize adc_seq here

// Queue up NUM_ASYNC asynchronous operations
for (int ai = 0; ai < NUM_ASYNC; ++ai) {
  k_poll_signal_init(adc_sig + ai);
  k_poll_event_init(adc_evt + ai, K_POLL_TYPE_SIGNAL,
                    K_POLL_MODE_NOTIFY_ONLY,
                    adc_sig + ai);
  u32_t now = k_cycle_get_32();
  rc = adc_read_async(adc, adc_seq + ai, adc_sig + ai);
  printk("Init %u got %d in %u us\n", ai, rc, 1000 * 1000 * (k_cycle_get_32() - now) / 32768);
}

// Wait for the operations to complete.
int ne = NUM_ASYNC;
do {
  u32_t now = k_cycle_get_32();
  rc = k_poll(adc_evt, ne, K_MSEC(1000));
  if (0 != rc) {
    printk("k_poll error: %d\n", rc);
    break;
  }
  int nc = 0;
  int ai = 0;
  while (ai < ne) {
    if (K_POLL_STATE_SIGNALED == adc_evt[ai].state) {
      ++nc;
      --ne;
      if (ai < ne) {
        memmove(adc_evt + ai, adc_evt + ai + 1, (ne - ai) * sizeof(*adc_evt));
      }
    } else {
      ++ai;
    }
  }
  printk("%u completed in %u us; %u rem\n", nc, 1000 * 1000 * (k_cycle_get_32() - now) / 32768, ne);
} while (0 < ne);
```

The output from the application is:
```
Init 0 got 0 in 0 us
Init 1 got 0 in 72113 us
Init 2 got 0 in 71716 us
Init 3 got 0 in 71716 us
3 completed in 30 us; 1 rem
1 completed in 69183 us; 0 rem
```

This confirms that the first operation is queued for asynchronous completion, but each of the subsequent ones cannot be queued until the previous operation completes.

I have not tested this on other platforms, but presumably they'd have the same problem.

The ADC asynchronous context implementation was cloned from the SPI implementation, which by inspection has exactly the same behavior, so I'm tagging that area too, though I haven't experimentally confirmed it's present in SPI.

If I'm mis-using this, please explain.  Blocking a thread that attempts to queue an operation because the resource is still in use by an earlier operation is really not what I want from an asynchronous API.

pabigot, 2019-04-26:
Linking in @andrewboie for comment:

The rationale in #15040 for copying the `adc_sequence` and `adc_options` objects into the async context instead of holding the sequence pointer appears to be undercut by the sequence object itself holding a pointer to the result buffer, which could equally be on the heap or call stack.   `struct spi_context` has exactly the same sort of design but with more pointers to external objects.

In any case, it's the fact that the device-unique async context holds the current operation's description by value that leads to the problem described in this issue.

I suspect the whole async context approach needs to be redesigned to be more like `struct _timeout` where the per-operation configuration is captured in a container linked through a `sys_dlist` held by the context.  If there are constraints on where that container or objects it points to may be located (static, heap, stack) they should be documented at the asynchronous API calls and/or the container structure itself.

tbursztyka, 2019-04-29:
Not a bug here. Looks like we have a misunderstanding in API usage: there is no queuing mechanism in SPI or ADC (anywhere in other API in fact), so it will not pile up calls. The async is on once the transaction is on. As long as the caller does not have its turn on the driver it will be blocked yes. Fact that there is no timeout is another issue, we brought this to the table many times as most of the low level device driver API would need that and it never came to any end.

If you want to be able to queue adc, spi (or any other type of buses) calls you will need a generic way to do it properly. 

Moving this to enhancement.

pabigot, 2019-04-29:
I don't particularly mind if `foo_async()` is restricted to a single in-progress operation.  I do mind that my thread is busy-waiting on a spin-lock for 72 ms.  The API (in my opinion) should return `-EINUSE` or `-EBUSY` if the resource is not available immediately, and the spinlock should not be held for the duration of the operation.

tbursztyka, 2019-04-29:
@pabigot we could do that, but only for async functions, others should still wait until they get their turn (actually there is no code using these async functions besides the loopback test, so it won't arm doing that change)

pabigot, 2019-04-29:
While I would find great value in an API that allowed me to queue multiple operations I fear the complexity of dealing with an intermediate operation that failed wouldn't be worth it.  So the remaining part of this issue is the (mis-)use of the spin-lock.  I've opened #15736 to cover that concern.

pabigot, 2019-04-29:
Somehow I'd gotten confused and thought the problem was a spin-lock being held; it's a semaphore.  So the change to the API to have it return `-EBUSY` or `-EAGAIN` if the resource isn't available would be a satisfactory resolution to this.

Looking at the code it's non-trivial to accomplish, so I'm withdrawing my offer to do this for ADC.