zephyrproject-rtos/zephyr #15706: tunslip6: main: open: No such file or directory

lorepieri8, 2019-04-27:
**Describe the bug**
Trying to follow https://docs.zephyrproject.org/1.13.0/subsystems/networking/qemu_setup.html in docker (Ubuntu 18.04) I fail to complete step 2 without errors.

**To Reproduce**
1. In terminal 2, ./loop-socat.sh
2. In terminal 3, sudo ./loop-slip-tap.sh

**Expected behavior**
Unknown. (Not specified in the link above)

**Screenshots or console output**
```
********SLIP started on ``/tmp/slip.dev''
slipfd and inslip reopened
ip neigh flush dev tap0
Cannot find device "tap0"
tunslip6: main: open: No such file or directory
```

**Environment (please complete the following information):**
 - OS: docker, Ubuntu 18.04

jukkar, 2019-05-01:
Instead of SLIP, you might want to try real Ethernet driver like e1000 in QEMU. See https://docs.zephyrproject.org/1.14.0/guides/networking/networking_with_host.html and  https://docs.zephyrproject.org/1.14.0/guides/networking/qemu_eth_setup.html#networking-with-eth-qemu for details.

This issue seems to be related to docker instead of zephyr, so removing the bug label.

nashif, 2019-05-07:
leave it as a bug please, if it is not a bug in zephyr, it should be closed instead or filed in the right place.

lorepieri8, 2019-05-08:
@jukkar 

On Digital Ocena I get

```
root@lorenzoPractice:~/zephyr_networking/net-tools# ./net-setup.sh
Using ./zeth.conf configuration file.
Creating zeth
RTNETLINK answers: File exists
```

while on Docker:

```
root@3e5b46fa1a8b:/net-tools# ./net-setup.sh 
Using ./zeth.conf configuration file.
Creating zeth
open: No such file or directory
Cannot find device "zeth"
Cannot find device "zeth"
Cannot find device "zeth"
Cannot find device "zeth"
Cannot find device "zeth"
```

jukkar, 2019-05-13:
Perhaps you have something incorrectly setup in the docker image so that the `ip tuntap add zeth mode tap` fails. Can you try to run it manually and see what errors it will print.
Closing this one for now as this does not seem to be related to issue in zephyr.

closed 2019-05-13 by jukkar

lorepieri8, 2019-05-17:
@jukkar 
It was indeed some networking capability missing in docker. Using

`docker run --cap-add=NET_ADMIN --device /dev/net/tun:/dev/net/tun  -it f69f013e46ef /bin/bash`

where `f69f013e46ef` is the CONTAINER ID will fix it.

ouaaliw, 2024-02-29:
ip tuntap add zeth mode tap
open: No such file or directory  
i am using WSL, i didn't understand this error. maybe because of the kernel limitation of WSL!!