zephyrproject-rtos/zephyr #15709: CODEOWNERS ignored in GitHub

aescolar, 2019-04-27:
**Describe the bug**
New PRs do not get reviewers automatically assigned thru the CODEOWNERS file, and CODEOWNER status is not displayed in the reviewer list

**To Reproduce**
a) Create a PR against code owned by anybody, see there is no automatic added reviewers
b) Note that in all PRs for the last couple of days there is no shield code owner symbol next to any reviewer.

**Impact**
No appropriate reviewers are added to PRs=>
* Likely that PRs will be forgotten
* Likely that inappropriate reviews will be done

**Additional context**

Discussion in devel and tsc mail list:
```
Email On Fri, 26 Apr 2019 09:25:17 +0000
"Alberto Escolar Piedras" wrote:

> Thanks Paul,
> Do you have some indication of when did this break?

Well, I was on vacation for two weeks before Apr 15. It definitely worked before that. And I saw first problematic case last week, after returning from vacation. No idea if that might be related to
https://github.com/zephyrproject-rtos/zephyr/commit/636a7af43e704e25af29b31d565399fc02fef408
- nothing wrong is seen here.

Actually, fishing for that link, I see that a whole bunch of changes was made to the CODEOWNERS file just recently:
https://github.com/zephyrproject-rtos/zephyr/commits/master/CODEOWNERS

Just imagine that any of those changes could break syntax a bit or propagation of changes to it could glitch on Github side, leading to the effect we see. And that's not counting a possible coincidence of Github tighting up the syntax for that file...

> 
> BR
> Alberto
> 
> -----Original Message-----
> From: tsc@lists.zephyrproject.org On Behalf Of Paul Sokolovsky
> Subject: [Zephyr-tsc] Github CODEOWNERS 
> auto-assignment doesn't work as expected
> 
> Hello,
> 
> As was noticed in a few (multiple?) PRs, automatic reviewer assignment 
> based on the CODEOWNERS file no longer works as expected, specifically 
> fails to assign reviewers.
> 
> A recent example of this is
> https://github.com/zephyrproject-rtos/zephyr/pull/15670 - while 
> there's an entry:
> 
> /subsys/net/ip/                           @jukkar @tbursztyka @pfalcon
> 
> in CODEOWNERS, I wasn't requested for review. Again, that's not the 
> only case.
> 
> For a clean-room test, I submitted
> https://github.com/zephyrproject-rtos/zephyr/pull/15683 and as can be 
> seen, the reviewer list is empty.
> 
> This is problematic, because there's a risk that under-reviewed 
> changes will get into the codebase, causing regressions. So, raising 
> to the awareness of the TSC.
```

aescolar, 2019-04-27:
CC @pfalcon

aescolar, 2019-05-03:
@nashif 
Would it make sense for an admin to raise this to GitHub support?
(Given that there does not seem to be anything obviously wrong with the CODEOWNERS file itself)

nashif, 2019-05-03:
looks like the issues start with 9e8ad2dfe77565a00ade186b157134397b50dd4c, need to double confirm, but I see no issue with the change itself, weird.

pfalcon, 2019-05-03:
>  9e8ad2d, need to double confirm, but I see no issue with the change itself, weird.

Well, it has commas in filenames.

nashif, 2019-05-03:
yep, I came to the same conclusion. The confusing part was that someone else added one more entry with commas which got me confused and was not able to get it to work by remove the entries from 9e8ad2d

closed 2019-05-03 by nashif

marc-hb, 2019-05-04:
> Would it make sense for an admin to raise this to GitHub support?

The feedback is "already in the right hands" : https://lists.zephyrproject.org/g/devel/message/5950 (2018-08-31)

wesleyolis, 2021-10-20:
It looks to be that, there is a hook into that it only shows the icon after the first pull request in which possible also a change to the codeowners file. Doing a setting change toggle doesn't work for branch protection rules.
You must make a change to the code owners file and the do a merge request, in which triggers a hook I imagine in GitHub app, after which the icon displays and it seem to work for me.

PR #15888 (2019-05-03): CODEOWNERS: minor cleanup
Remove redundant lines and minor owner updates.

Fixes #15709