zephyrproject-rtos/zephyr #15710: [question] how about the current consumption on NRF52DK running power_mgr sample?

summer20100514, 2019-04-28:
It seems that power subsystem has been updated during 1.14 release. As old issues #12150 and #12367 indicate the current consumption is not controlled well. So what's the newest current result and if none, what's the roadmap on this topic? thank you.
@pizi-nordic

pizi-nordic, 2019-04-29:
Key facts:
1. At the moment there is no comprehensive power management solution in Zephyr. There are 2 distinct and limited frameworks:
   - System Power Manager - which chooses power mode depending on the expected idling time.
   - Device Idle Power Manager - which disables devices which are not used.
2. The frameworks mentioned above does not cooperate witch each other, so for example devices are not automatically put in low power state when the system is.
3. Due to recent introduction of the device power management not all drivers are implementing the required API or such implementation has a flaws. As result power consumption is excessive in many cases.

Roadmap: We would like to introduce real and comprehensive power management, however we are still far from the goal. What we see is lack of common mechanism needed for such framework (deltails: https://github.com/zephyrproject-rtos/zephyr/issues/14307) and we are now trying to fill the gaps.

summer20100514, 2019-04-29:
@pizi-nordic thank you.

closed 2019-04-29 by summer20100514

endian-benjamin, 2019-05-03:
Keyer fact: the Nordic drivers pm implementations are all utterly broken.

chris-makaio, 2019-06-04:
I can confirm, @endian-benjamin 
But... uhm... how to fix that?
My log gets spammed with 

> [00:00:07.444,519] <err> soc: Unsupported power state 0
> [00:00:07.444,519] <err> soc: Unsupported power state 0
> [00:00:07.447,601] <err> power: Not enough time for PM operations: 1
> [00:00:07.453,125] <err> power: Not enough time for PM operations: 1

endian-benjamin, 2019-06-04:
I had a fix for that particular nuisance, but must have forgot to push it.That shouldn't be logged as an error as it's expected. Will push a fix.

We've actually managed to get power management on nRF52 working pretty decently after struggling with it for a long time. The power API isn't clear at all, the sample isn't clear at all and the drivers were in poor shape. I could try to help write up some sample/documentation/guide for PM, but alas it seems no one wants to come near the power subsystem right now. Everyone agrees that it needs to be addressed but crucial people are busy with other things ATM.

chris-makaio, 2019-06-04:
well, I'm currently tinkering with the PM but really, it's a mess. Could you maybe push your PR (or let me know your commit in your own repo)? Would be really nice...

endian-benjamin, 2019-06-04:
I'll see what I can do soonish, but I don't have time right now.

endian-benjamin, 2019-06-04:
Pull request submitted, but in the meanwhile you cant rest assured that those aren't actually errors at all and shouldn't be logged as such.

And for the record, don't expect to do much power saving with `CONFIG_LOG=y`, because the logging subsystem keeps a thread that wakes the system up at 10 Hz or so. You can configure this with `CONFIG_LOG_IMMEDIATE` but it's pretty shaky - if you want to verify power savings, just stick to current meters and, in general, non-console debugging.

endian-benjamin, 2019-06-04:
...and remember that both JLink RTTLogger and GDBServer can squeeze around 1.6 mA or so of consumption.

endian-benjamin, 2019-06-04:
...and also note that the power_mgr sample is a very poor example of actual power management. It doesn't actually attempt to SAVE POWER, it just tries to give a VERY basic overview of the APIs.

chris-makaio, 2019-06-04:
Thx!
Well, I somewhat need the log to figure causes for not going to sleep... pretty hard.
I worked on my code, disabled UART and made some other config changes (I can post them if required) but I feeling really bad not knowing what I‘m doing.
For example, I‘m now down from 6mA idle current to 1mA (which is fine because of other peripherals on the board).
But as soon as I disable logging (incl shell, console log subsys and rtt) it goes up to 4mA. Wtf?!
Same for min residencies for PM: I would expect more savings when using lower thresholds (100ms instead of 5000ms for example), but again, current goes up.

It‘s not an option for me to disable UART during compile time, so I tried device PM which obviously doesn‘t work either...

endian-benjamin, 2019-06-04:
If you lower thresholds, the device will constantly (i mean CONSTANTLY) go in and out of power modes, which costs a lot of cpu cycles. You won't save ANY power with those low thresholds.

endian-benjamin, 2019-06-04:
Also, the log IS the cause for not going to sleep... but if you really need it, try config LOG_IMMEDIATE