zephyrproject-rtos/zephyr #15713: MCUMGR_LOG build error

lgl88911, 2019-04-28:
**Describe the bug**
when I config CONFIG_MCUMGR_CMD_LOG_MGMT=y,  mcumgr log will build error:

**error 1**: create makefile fail:
``` 
Cannot find source file:
/home/frank/work/project/westzephyr/zephyr/ext/lib/mgmt/mcumgr/cmd/log_mgmt/cmd/log_mgmt/port/zephyr/src/zephyr_log_mgmt.c
```
I fix it by modify *zephyr/ext/lib/mgmt/mcumgr/cmd/log_mgmt/CMakeLists.txt*
```
target_include_directories(MCUMGR INTERFACE 
    include
)

zephyr_library_sources(
    port/zephyr/src/zephyr_log_mgmt.c
    src/log_mgmt.c
    src/stubs.c
)
```
then disapper
**error 2** can't find file:
```
/home/frank/work/project/westzephyr/zephyr/ext/lib/mgmt/mcumgr/cmd/log_mgmt/port/zephyr/src/zephyr_log_mgmt.c:21:10: fatal error: logging/mdlog.h: No such file or directory
 #include <logging/mdlog.h>
```

I search zephyr path & west path， can't find mdlog， but it be used by zephyr_log_mgmt.c.
It's a bug? or I config have some error?
Please give me some suggest, thanks!

carlescufi, 2019-04-28:
That file doesn't seem to have ever existed. It might be that this has never been correctly tested. @ccollins476ad can you shed some light?

carlescufi, 2019-05-14:
Relabeled as enhancement since this has never been actually supported. Contributors welcome to address this.

closed 2020-10-05 by sjanc