zephyrproject-rtos/zephyr #15720: "z_clock_elapsed" implementation seems to be missing #linking #sched

StefJar, 2019-04-29:
After pulling the latest Zephyr code(8bd28db98035f1bb4b66ae8956810c5b2041a16d) and running the build of my project, the linking fails.

I am running Zephyr on a stm32f4(custom board).

The linking fails with:
```
make -j4
[  1%] Built target kobj_types_h_target
[  1%] Built target driver_validation_h_target
[  2%] Built target syscall_macros_h_target
[  2%] Generating misc/generated/syscalls.json
[  4%] Built target syscall_list_h_target
[  5%] Built target offsets
[  5%] Built target offsets_h
[  5%] Built target linker_script_target
[ 13%] Built target arch__arm__core
[ 27%] Built target app
[ 41%] Built target kernel
[ 43%] Built target arch__arm__core__cortex_m__mpu
[ 50%] Built target lib__libc__minimal
[ 54%] Built target arch__arm__core__cortex_m
[ 55%] Built target ..__git__khpfw__boards__arm__khd1_hb_v1_4a
[ 56%] Built target drivers__gpio
[ 89%] Built target zephyr
[ 90%] Built target drivers__pwm
[ 92%] Built target drivers__i2c
[ 93%] Built target drivers__spi
[ 94%] Built target drivers__serial
[ 95%] Linking C executable zephyr_prebuilt.elf8bd28db98035f1bb4b66ae8956810c5b2041a16d
Memory region         Used Size  Region Size  %age Used
           FLASH:       94120 B         1 MB      8.98%
            SRAM:       39684 B       256 KB     15.14%
        IDT_LIST:         296 B         2 KB     14.45%
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: kernel/libkernel.a(sched.c.obj): in function `reset_time_slice':
/home/stefan/zephyr/kernel/sched.c:258: undefined reference to `z_clock_elapsed'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: kernel/libkernel.a(thread.c.obj): in function `z_impl_k_busy_wait':
/home/stefan/zephyr/kernel/thread.c:111: undefined reference to `z_timer_cycle_get_32'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: /home/stefan/zephyr/kernel/thread.c:114: undefined reference to `z_timer_cycle_get_32'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: kernel/libkernel.a(timeout.c.obj): in function `elapsed':
/home/stefan/zephyr/kernel/timeout.c:58: undefined reference to `z_clock_elapsed'
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/8.3.0/../../../../arm-zephyr-eabi/bin/ld: kernel/libkernel.a(timeout.c.obj): in function `z_tick_get':
/home/stefan/zephyr/kernel/timeout.c:223: undefined reference to `z_clock_elapsed'
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:100: zephyr/zephyr_prebuilt.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:530: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
```
For me it looks like that the "z_clock_elapsed" function got lost.

ioannisg, 2019-04-29:
Yep, most likely due to #15212.
If this is a custom board, you, now, need to select the SYSTICK timer explicitly in your BOARD definition.

ioannisg, 2019-04-29:
@StefJar can you confirm that defining CONFIG_CORTEX_M_SYSTICK=y in BOARD defconfig solves your linking error?

StefJar, 2019-04-29:
@ioannisg I can confirm that this solves that linking error

ioannisg, 2019-04-29:
> 
> 
> @ioannisg I can confirm that this solves that linking error

Would you like to close this bug-report, then?

closed 2019-04-29 by StefJar

snegaa804, 2024-07-01:
hi [ioannisg](https://github.com/ioannisg) 

i am facing the similar issue i have tried the command u gave but still the same

[96/101] Linking C executable zephyr/zephyr_pre0.elf
FAILED: zephyr/zephyr_pre0.elf zephyr/zephyr_pre0.map /home/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr_pre0.map 
: && ccache /home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc  -gdwarf-4 zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj -o zephyr/zephyr_pre0.elf  zephyr/CMakeFiles/offsets.dir/./arch/arm/core/offsets/offsets.c.obj  -fuse-ld=bfd  -T  zephyr/linker_zephyr_pre0.cmd  -Wl,-Map=/home/zephyrproject/zephyr/samples/hello_world/build/zephyr/zephyr_pre0.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/arm/core/libarch__arm__core.a  zephyr/arch/arch/arm/core/cortex_m/libarch__arm__core__cortex_m.a  zephyr/lib/libc/picolibc/liblib__libc__picolibc.a  zephyr/lib/libc/common/liblib__libc__common.a  zephyr/drivers/timer/libdrivers__timer.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  -L"/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/thumb/v6-m/nofp"  -L/home/zephyrproject/zephyr/samples/hello_world/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -mcpu=cortex-m0plus  -mthumb  -mabi=aapcs  -mfp16-format=ieee  -mtp=soft  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -Wl,-no-pie  -DPICOLIBC_LONG_LONG_PRINTF_SCANF  --specs=picolibc.specs  -lc  -lgcc && cd /home/zephyrproject/zephyr/samples/hello_world/build/zephyr && /usr/bin/cmake -E true
/home/zephyr-sdk-0.16.8/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: zephyr/kernel/libkernel.a(timeout.c.obj): in function `elapsed':
/home/zephyrproject/zephyr/kernel/timeout.c:80: undefined reference to `sys_clock_elapsed'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.