zephyrproject-rtos/zephyr #15729: drivers: flash: should write_protection be emulated?

nvlsianpu, 2019-04-29:
According to https://docs.zephyrproject.org/latest/reference/peripherals/flash.html#_CPPv326flash_write_protection_setP6deviceb
write protection might behave different depending on which flash driver is used. Is this possible to relax it completely - so make this API no-operation for a driver. Then this driver would take care on its-own regard its internal protection.

I'm asking because for nRF flash driver write protection is completely emulated already - which will be not necessary if can relax it.

carlescufi, 2019-06-04:
Proposal:

- Modify the doc to reflect that this is a NOP on platforms that do not have hardware write protection
- Return 0 on those platforms that either do not have hardware write protection or it doesn't fit the API

d3zd3z, 2019-12-03:
What about flash drivers that don't ever need to write to the flash. For example, some flash drivers, writes are entirely done through the controller. I would assume these would just never call this API.

PR #17498 (2019-12-15): drivers/flash: Protection API clarification
### General patch
This changes was discuses on dev-meeting 4th June 2019

On some targets hardware (or middelware) doesn't allow
implement functionality flash protection API -
so fare it have to be emulated by software on such a target.

This patch changes documentation of this API, so on such a targets
API might implements no-operation.

fixes #15729

### Nordic patch

NRF devices hardware has flash protection which doesn't
reflect flash API definition well. So fare protection
mechanism was emulated by the software and the driver deals
with hardware [... cut]
commit aa0d41f7 (2019-12-15): drivers/flash: protection API clarification

On some targets hardware (or middelware) doesn't allow
implement functionality flash protection API -
so fare it have to be emulated by software on such a target.

This patch changes documentation of this API, so on such a targets
API might implements no-operation.

fixes #15729
closed 2019-12-15 by nashif