zephyrproject-rtos/zephyr #15736: Generalize and improve async context for SPI, ADC, etc.

pabigot, 2019-04-29:
Referred from #15703:

Spin-locks are intended for cases where the lock is expected to be released almost immediately, where the cost of suspending the calling thread is too high.  The SPI (ADC) async context uses a spin-lock to protect data from multiple access, but holds it while an operation is in progress.

Whether this is a bug or intended, it's very much not desirable.

A case might be made that the original use of the context design in SPI would never block for long enough to cause a problem.  For ADC it is very much incorrect, because it's easy to come up with ADC configurations on nRF52 that require over 500 ms to complete (8 channels at 256 oversamping 14 bit).  If the SAADC module is busy, having another thread spin for a half a second is unacceptable.

The current API `foo_async` behavior for SPI and ADC does not allow for a timeout, nor for the call to immediately fail if the resource is not available.  The simplest approach would be to make an async request return `-EBUSY` or `-EWOULDBLOCK` immediately if the resource is already in use.  https://github.com/zephyrproject-rtos/zephyr/issues/15703#issuecomment-487561436 suggests this might be acceptable.

However, it can't be done as long as the spin lock protecting the in-use state continues to be held until the device is available.

We already have two users of this async context capability (ADC and SPI) and it's likely others could benefit as well (e.g. asynchronous support in I2C).  I think this infrastructure should be replaced with a generic async context state that protects a resource with these features:
* a spin-lock that protects the state so calls can quickly succeed or fail if the resource is not available, but is not held while the device is active;
* a wait queue for any synchronous invocations that can't be immediately started.

Here I assume that the synchronous invocations can support suspending the thread, even though the current implementation doesn't do that.  If they can't, then I don't know what to do.

Even this isn't quite right, because it allows for interleaved use of the device from different threads while for the existing ADC API in particular it can be important to ensure that multiple operations are executed in sequence without interruption.  So there ought to be an option to take ownership of the device, where non-owner attempts to use are held or rejected until the owner releases it.

At any rate, just some thoughts.  I'll work around the existing behavior in other ways.

tbursztyka, 2019-04-29:
@pabigot btw, you are mentioning spin lock, but spi uses a semaphore, so I guess semaphores now uses a spin lock below. k_sem_take used to suspend the thread before. It does not do that anymore?

pabigot, 2019-04-29:
@tbursztyka You're right.  I have no idea why I kept reading that as a spin lock.

Never mind.  I'll switch to `-EWOULDBLOCK` for ADC in a patch in my fork; if it looks sane I'll upstream it and somebody else can do the same for SPI.

closed 2019-04-29 by pabigot