zephyrproject-rtos/zephyr #15738: Networking with QEMU for mac

lorepieri8, 2019-04-29:
**Is your enhancement proposal related to a problem? Please describe.**
Is there a mac version of Networking with QEMU? 
https://docs.zephyrproject.org/latest/guides/networking/qemu_setup.html#networking-with-qemu

carlescufi, 2019-04-30:
You can just install  QEMU with brew and it should work

lorepieri8, 2019-04-30:
The problem is that 

```
git clone https://github.com/zephyrproject-rtos/net-tools
cd net-tools
make
```

returns 

```
 lorenzo@MacBook-Air-di-Lorenzo ~/gitHubRepo/net-tools $ make
cc    -c -o tunslip6.o tunslip6.c
cc -o tunslip6   tunslip6.c
cc    -c -o echo-client.o echo-client.c
echo-client.c:27:10: fatal error: 'linux/sockios.h' file not found
#include <linux/sockios.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
make: *** [echo-client.o] Error 1
```

jukkar, 2019-05-01:
The tunslip6 program works only for Linux. Currently there is no plan to support Mac. Patches are welcome if you want to contribute Mac support.

d3zd3z, 2019-12-03:
Have we considered trying to find if there is a NIC that Qemu is able to emulate, that has a driver for Zephyr? In that configuration, we'd had a lot more options, and the networking would be managed by Qemu, say through tun/tap or equivalent mechanisms on Mac.

jukkar, 2019-12-04:
For x86, the Qemu supports Intel e1000 network card and we have a driver for that in Zephyr. For ARM based Qemu, there is for example stellaris ethernet driver support in Zephyr.
I do not have a mac, so cannot really comment how much changes would be needed in host side setup script. If you want to investigate this, the net-tools project has `net-setup.sh` script that is used to setup the host side of the tun device. The relevant Qemu configuration in Zephyr side is done by `cmake/app/boilerplate.cmake` script (search for `zeth` string in that file)

beriberikix, 2021-03-14:
FYI, there's an [early patch](https://mail.gnu.org/archive/html/qemu-devel/2021-02/msg04637.html) for QEMU that uses the modern native network emulation features in macOS.

beriberikix, 2021-03-16:
I've been testing the patch from @codyd51 and it seems to correctly create network interfaces 🎉 ! I'm at the point where someone more experienced with Zephyr might be needed.  My particular test is with `qemu_cortex_m3` (since [x86 doesn't work](https://github.com/zephyrproject-rtos/zephyr/issues/10945#issuecomment-751499639) on macOS) and after I build with an echo test, here's what I'm seeing:

```
jonathanberi:zephyr 🚀 $ sudo /Users/jonathanberi/downloads/qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -vga none -netdev vmnet-macos,id=vmnet,mode=bridged -device rtl8139,netdev=vmnet -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel /Users/jonathanberi/zephyrproject/zephyr/build/zephyr/zephyr.elf
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: Started vmnet interface with configuration:
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: MTU:              1500
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: Max packet size:  1514
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: MAC:              ba:8c:3a:8a:fd:ee
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: UUID:             4B84C3D0-BDBE-405C-888E-BD04740BE69F
qemu-system-arm: -device rtl8139,netdev=vmnet: No 'PCI' bus found for device 'rtl8139'
jonathanberi:zephyr 🚀 $ sudo /Users/jonathanberi/downloads/qemu-system-arm -cpu cortex-m3 -machine lm3s6965evb -nographic -vga none -netdev vmnet-macos,id=vmnet,mode=bridged -device stellaris,netdev=vmnet -pidfile qemu.pid -chardev stdio,id=con,mux=on -serial chardev:con -mon chardev=con,mode=readline -kernel /Users/jonathanberi/zephyrproject/zephyr/build/zephyr/zephyr.elf
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: Started vmnet interface with configuration:
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: MTU:              1500
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: Max packet size:  1514
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: MAC:              6a:31:4e:7a:8d:8e
qemu-system-arm: -netdev vmnet-macos,id=vmnet,mode=bridged: info: UUID:             DCBC030E-4687-452C-8880-5443E75216A4
qemu-system-arm: -device stellaris,netdev=vmnet: 'stellaris' is not a valid device model name
```

Anyone willing and able to work on this with me? 🙏

closed 2021-05-20 by nashif