zephyrproject-rtos/zephyr #15748: 'ninja flash' does not work for IMXRT1052 target

xiongyu0523, 2019-04-30:
**Describe the bug**
Follow the guide to run 'ninja flash', all build processes are passed just stop the last step. It shows 'FAILED: zephyr/cmake/flash/CMakeFiles/flash'

**To Reproduce**
Steps to reproduce the behavior:
1. Install Jlink OpenSDA firmware for the on-board debugger
2. The debugger itself works well, e.g. using J-Link Lite to program a zephyr.bin binary is good.
3. Go to samples/hello_world

```
mkdir build & cd build
cmake -GNinja -DBOARD=mimxrt1050_evk
ninja flash
```

Full error message

```
Using runner: jlink
Flashing Target Device
Traceback (most recent call last):
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\nxa07532\AppData\Local\Programs\Python\Python37-32\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\site-packages\west\_bootstrap\main.py", line 499, in main
    wrap(wrap_argv)
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\site-packages\west\_bootstrap\main.py", line 485, in wrap
    west.main.main(argv)
  File "C:\Users\nxa07532\zephyrproject\.west\west\src\west\main.py", line 580, in main
    args.handler(args, unknown)
  File "C:\Users\nxa07532\zephyrproject\.west\west\src\west\main.py", line 332, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "C:\Users\nxa07532\zephyrproject\.west\west\src\west\commands\command.py", line 90, in run
    self.do_run(args, unknown)
  File "c:\users\nxa07532\zephyrproject\zephyr\scripts\west_commands\flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 228, in do_run_common
    runner.run(command_name)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 407, in run
    self.do_run(command, **kwargs)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\jlink.py", line 99, in do_run
    self.flash(**kwargs)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\jlink.py", line 150, in flash
    self.check_call(cmd)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 466, in check_call
    subprocess.check_call(cmd)
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\subprocess.py", line 342, in check_call
    retcode = call(*popenargs, **kwargs)
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "c:\users\nxa07532\appdata\local\programs\python\python37-32\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
**FileNotFoundError: [WinError 2] The system cannot find the file specified
FAILED: zephyr/cmake/flash/CMakeFiles/flash**
cmd.exe /C "cd /D C:\Users\nxa07532\app\build && "C:\Program Files\CMake\bin\cmake.exe" -E env C:/Users/nxa07532/AppData/Local/Programs/Python/Python37-32/Scripts/west.exe flash --skip-rebuild"
ninja: build stopped: subcommand failed.
```

aescolar, 2019-04-30:
CC @mbolivar

mbolivar, 2019-04-30:
@xiongyu0523 can you please run `west -vvv flash` from the build directory? That will provide more verbose debug information than plain `ninja flash`. It may help @MaureenHelm diagnose.

xiongyu0523, 2019-05-01:
Log after run `west -vvv flash`, it seems i do not have commander file in below path C:\Users\nxa07532\AppData\Local\Temp\1\tmpk6ydop3ijlink\runner.jlink

Previous I have two python37 installed, one 32-bit version and one 64-bit version. Now I uninstalled 32-bit in AppData\Programs, only keep the python37 in c:\ root directory. I'm not sure whether it impacts the script generation.

```
Using runner: jlink
JLink commander script:
r
loadfile C:/Users/nxa07532/app/build/zephyr/zephyr.bin 0x60000000
g
q
Flashing Target Device
JLink.exe -if swd -speed auto -device MCIMXRT1052 -CommanderScript 'C:\Users\nxa07532\AppData\Local\Temp\1\tmpk6ydop3ijlink\runner.jlink'
Traceback (most recent call last):
  File "c:\python37\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "c:\python37\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Python37\Scripts\west.exe\__main__.py", line 9, in <module>
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 499, in main
    wrap(wrap_argv)
  File "c:\python37\lib\site-packages\west\_bootstrap\main.py", line 485, in wrap
    west.main.main(argv)
  File "C:\Users\nxa07532\zephyrproject\.west\west\src\west\main.py", line 580, in main
    args.handler(args, unknown)
  File "C:\Users\nxa07532\zephyrproject\.west\west\src\west\main.py", line 332, in ext_command_handler
    command.run(*west_parser.parse_known_args(argv))
  File "C:\Users\nxa07532\zephyrproject\.west\west\src\west\commands\command.py", line 90, in run
    self.do_run(args, unknown)
  File "c:\users\nxa07532\zephyrproject\zephyr\scripts\west_commands\flash.py", line 32, in do_run
    'ZEPHYR_BOARD_FLASH_RUNNER')
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\run_common.py", line 228, in do_run_common
    runner.run(command_name)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 407, in run
    self.do_run(command, **kwargs)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\jlink.py", line 99, in do_run
    self.flash(**kwargs)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\jlink.py", line 150, in flash
    self.check_call(cmd)
  File "C:\Users\nxa07532\zephyrproject\zephyr\scripts/west_commands\runners\core.py", line 466, in check_call
    subprocess.check_call(cmd)
  File "c:\python37\lib\subprocess.py", line 342, in check_call
    retcode = call(*popenargs, **kwargs)
  File "c:\python37\lib\subprocess.py", line 323, in call
    with Popen(*popenargs, **kwargs) as p:
  File "c:\python37\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "c:\python37\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified
```

MaureenHelm, 2019-05-01:
It looks like you need to add the jlink commander to your windows search path. You may want to create a `zephyrrc.cmd` file in your home directory to do this automatically. It gets invoked when you run `zephyr-env.cmd`.

Here's mine:
```
set ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
set GNUARMEMB_TOOLCHAIN_PATH=c:\gnuarmemb\gcc-arm-none-eabi-7-2018-q2

set PATH=%PATH%;C:\Program Files (x86)\SEGGER\JLink_V634h
```

carlescufi, 2019-05-02:
Closing as this is a known issue that has been reported several times

closed 2019-05-02 by carlescufi

xiongyu0523, 2019-05-02:
Indeed, everything is fine after adding to PATH. The debug message does not provide clear information for this error. Better to document it in 'J-Link Debug Host Tools' seciton of documents.  Thank you

PR #15829 (2019-05-03): scripts: west_commands: Print error message when jlink exe not found
Adds a check in the jlink runner to look for the jlink executables and
print a more useful error message if they are not found.

Fixes #15748

```
C:\Users\nxa17364\zephyrproject\zephyr\samples\hello_world>west flash
ninja: no work to do.
Using runner: jlink
FATAL ERROR: JLink.exe is not installed or cannot be found

C:\Users\nxa17364\zephyrproject\zephyr\samples\hello_world>west debug
ninja: no work to do.
Using runner: jlink
FATAL ERROR: JLinkGDBServer is not installed or cannot be found
```