zephyrproject-rtos/zephyr #15749: [question] errors using custom command in CMakeLists.txt

summer20100514, 2019-04-30:
can you point me out why this command won't wok which resides in my application CMakelists.txt? thank you.
```add_custom_command(TARGET ${ZEPHYR_RUNNER_CONFIG_KERNEL_HEX} POST_BUILD
    COMMAND
    ${PYTHON_EXECUTABLE}
    ../../mesh-scripts/test_.py ${APPLICATION_BINARY_DIR} ${ZEPHYR_RUNNER_CONFIG_KERNEL_HEX}
)```
The output is
```CMake Error at CMakeLists.txt:43 (add_custom_command):
  No TARGET
  'D:/Software_Related/server/bulb/build/zephyr/zephyr.hex'
  has been created in this directory.```

summer20100514, 2019-05-06:
no updates?

SebastianBoe, 2019-05-06:
add_custom_command(TARGET 

must be used with a target, but a file has been specified.

What you would want to do to get a POST_BUILD command is to set the property

extra_post_build_commands

from before boilerplate.cmake.

The only problem being that you don't know what

 ${PYTHON_EXECUTABLE}
 ${APPLICATION_BINARY_DIR}
${ZEPHYR_RUNNER_CONFIG_KERNEL_HEX}

these evaluate to until after boilerplate.cmake.

It might also be possible to create a custom target is a part of ALL and that depends on the target ${logical_target_for_zephyr_elf}.

closed 2019-05-06 by SebastianBoe

summer20100514, 2019-05-06:
problem solved, thank you! @SebastianBoe @carlescufi