zephyrproject-rtos/zephyr #15751: Incorrect flash map

PhilippFinke, 2019-04-30:
Hi everybody,

it seems I found a bug in v1.14.0. I'm using Zephyr on my nrf52840 based custom board together with mcuboot v1.3.0. Everything worked fine with Zephyr v1.13.0, but after updating to the latest release I'm not able to boot the image in slot 0 via mcuboot. After debugging for a while, I had a look at the array flash_map (pointing to default_flash_map). This array has 5 entries that reflect the partitions defined in the board's dts, except that the entries for slot 0 and 1 are turned around. Because of this mcuboot is pointing to slot 1 when it wants to boot from slot 0 and vice versa. I first thought this is an issue with my custom board, but I have the same problem building for e.g. nrf52840_pca10056.

Did I simply miss something here or is this really an issue?

Best

 

Philipp

nvlsianpu, 2019-04-30:
https://github.com/JuulLabs-OSS/mcuboot/issues/429

PhilippFinke, 2019-04-30:
I don't think that this is a solution to my problem. I can build mcuboot v1.3.0 against Zephyr v1.14.0. The problem is that the flash map is incorrectly arranged. The problem even exists when I build a standalone version of my application. The array default_flash_map (that is used by mcuboot to retrieve the slot information) looks like this in the debugger.

default_flash_map[0] = mcuboot partition info
default_flash_map[1] = slot 1 partition info
default_flash_map[2] = slot 0 partition info
default_flash_map[3] = scratch partition info
default_flash_map[4] = storage partition info

mcuboot expects the info of slot 0 on array index 1, but there is the info of slot 1.

carlescufi, 2019-05-03:
@lemrey @mbolivar can you take a look since @nvlsianpu is currently on holiday?

PhilippFinke, 2019-05-06:
Using the mcuboot master everything works now.

closed 2019-05-06 by nvlsianpu