zephyrproject-rtos/zephyr #15759: usb: cdc_acm: uart_line_ctrl_set(dev, LINE_CTRL_DTR, &dtr) should always return 0 if USB port is not opened by host

jli157, 2019-04-30:
**Describe the bug**
A clear and concise description of what the bug is.
The status of LINE_CTRL_DTR retrieved by the function `uart_line_ctrl_set` is useful to detect if the program running on the host computer already opened the USB CDC-ACM port or not. However, it is observed that the status of LINE_CTRL_DTR is not consistent. 

**To Reproduce**
Steps to reproduce the behavior:
- board booting up 
- `DTR = 0` for about 10 - 20 seconds 
- **DTR changed to `1` for about 10 seconds even though the peer program doesn't open the USB port**
- DTR changed back to `0` 
- DTR = 1 when the peer program opens USB port 

**Expected behavior**
Before the peer program opens CDC-ACM USB port, the status of DTR should be always zero. 

**Impact**
The inconsistence of DTR status results in mis-judgement on if the peer program is running or not. 

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: Ubuntu 16.04
 - Toolchain Zephyr SDK 1.10.0
 - Commit SHA or Version used: zephyr@4b2a181 4/30/2019

**Additional context**
Add any other context about the problem here.

jli157, 2019-04-30:
@finikorg please help to analyze the issue, thank you!

finikorg, 2019-05-02:
Do you have logs? DTR can be change when you see `usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE`.
BTW do you have modem manager or some other apps accessing port?

jli157, 2019-05-02:
> Do you have logs? DTR can be change when you see `usb_cdc_acm.cdc_acm_class_handle_req: CDC_SET_CONTROL_LINE_STATE`.
> BTW do you have modem manager or some other apps accessing port?

I have a function 
```
static bool serial_io_dtr_signaled(struct serial_context *desc)
{       
        if (desc->config->enable_tx_irq) {
#ifdef CONFIG_UART_LINE_CTRL
                u32_t dtr = 0;
                uart_line_ctrl_get(desc->port_dev, LINE_CTRL_DTR, &dtr);
                return dtr;
#endif
        }
 
        return true;
}
```
which is called by a high-level writing function to check if the other peer is ready. If the peer is not ready, the writing operation will be ignored. 

If the function returns zero, a printk("DTR\n") will be called. Otherwise, how many written bytes will be printed out. 

The upper level writing function will be called by a sensor data collection thread every like 20ms, to output sensor data to the host. 

Actually, if USB stack is able to deal with writing operation without the peer opening the port, I don't think this projection is necessary. The current problem is the USB stack will get stuck before the peer starts to run. So, I have to use this DTR detection function to protect it. However, inconsistent DTR behavior results in writing operations unprotected. 

The log file is attached, in which the line "DTR" means DTR = 0 while the line "w:xx" means how many bytes were written to the USB stack. You can see at the beginning of booting up, DTR = 0 as expected (actually at this moment, the USB port can't be opened as well); after a while DTR becomes true and writing operations are conducted (those lines like "w:xx") which are not expected; then DTR becomes zero again after about 5-10 seconds. 

[dtr_debug.txt](https://github.com/zephyrproject-rtos/zephyr/files/3138565/dtr_debug.txt)

jli157, 2019-05-02:
hmm, it seems Modem manager is running on my host computer. It could be culprit of the issue. I'll give it another try today by disabling modem manager firstly. 

```bash
$ systemctl | grep Modem
ModemManager.service             loaded active running   Modem Manager

```

jli157, 2019-05-02:
@finikorg Okay, after disabling Modem Manger, I didn't see DTR becomes 1 now. So, it is Modem Manager which set DTR=1 for a while, I believe.

jli157, 2019-05-02:
It turns out the modem manager on the host opened the port for a while and made it inconsistent. So, it is not an issue from the usb stack.

closed 2019-05-02 by jli157