zephyrproject-rtos/zephyr #15769: [Coverity CID :198012]Memory - corruptions in /drivers/flash/flash_simulator.c

aasthagr, 2019-05-01:
Static code scan issues seen in File: /drivers/flash/flash_simulator.c
 Category: Memory - corruptions
 Function: flash_init
 Component: Drivers
 CID: 198012
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

ioannisg, 2019-06-17:
Any updates on this @lemrey @nvlsianpu?

nvlsianpu, 2019-06-17:
might be not a issue anymore by https://github.com/zephyrproject-rtos/zephyr/issues/16572
I would wiat for next coverity scan report.

ioannisg, 2019-06-25:
Is this one still an issue @lemrey ??

lemrey, 2019-06-28:
This was fixed in https://github.com/zephyrproject-rtos/zephyr/pull/16635.
It was actually the in the `stats` subsystem :)

ioannisg, 2019-06-28:
@lemrey if this is fixed, could you please, close the issue?

closed 2019-06-28 by nvlsianpu