zephyrproject-rtos/zephyr #15793: Unable to load binaries into iotdk

nashif, 2019-05-01:
**Describe the bug**

Following the instructions for the iotdk, I am not able to load any binaries to the device.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b iotdk samples/synchronization/
2. west flash

```
i9:zephyr(master): west flash
ninja: no work to do.
Using runner: em-starterkit
Open On-Chip Debugger 0.10.0+dev-g0b1cbf175-dirty (2019-03-31-07:39)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 8000 kHz
Info : clock speed 8000 kHz
Info : JTAG tap: arc-em.cpu tap/device found: 0x200444b1 (mfg: 0x258 (ARC International), part: 0x0044, ver: 0x2)
Register `identity' is not found.

Info : Listening on port 3333 for gdb connections
    TargetName         Type       Endian TapName            State
--  ------------------ ---------- ------ ------------------ ------------
 0* arc-em.cpu         arcv2      little arc-em.cpu         halted
Info : Listening on port 6333 for tcl connections
Info : Listening on port 4444 for telnet connections
GNU gdb (crosstool-NG 1.24.0-rc2-dirty) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=arc-zephyr-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/nashif/Work/zephyrproject/zephyr/build/zephyr/zephyr.elf...done.
Remote debugging using :3333
Info : accepting 'gdb' connection on tcp/3333
warning: while parsing target description (at line 6): Boolean fields must be one bit in size
warning: Could not load XML target description; ignoring
openocd: src/server/gdb_server.c:1188: gdb_get_registers_packet: Assertion `reg_packet_size > 0' failed.
Remote connection closed
You can't do that when your target is `exec'
"monitor" command not supported by this target.

```

**Expected behavior**

Binary should be loaded 

**Impact**
Showstopper

**Additional context**

How can we flash a binary into the internal flash?

ruuddw, 2019-05-02:
See [SDK issue 65](https://github.com/zephyrproject-rtos/sdk-ng/issues/65), I believe this is the same issue.
Workaround is using an older openocd binary from a previous Zephyr SDK (I believe Zephyr SDK 0.9.3 was still ok).

ARC development boards typically download binaries to ICCM, I don't believe a flash runner has been added yet. @vonhust do you have a pointer on how to use the flash, or even better a plan for a flash runner?

nashif, 2019-05-03:
@ruuddw thanks for confirming, I suspected it was that issue.
@vonhust any hints on flashing to ICCM? What is the method of getting the binary into the flash?

vonhust, 2019-05-05:
@nashif  this issue is caused by ARC's openocd support as pointed out by Ruud.
I just used the openocd in ARC GNU toolchain to replace Zephyr SDK's openocd to debug.
If openocd can run,  the binray/elf file can be directly written into CCMs

@ruuddw @nashif ,  iotdk has the following flash
* internal eflash  (no detailed datasheet, but several simple register description)
* 2 MB spi flash connected to internal bootspi controller with XIP support (no detailed datasheet, but several simple register description)
* 2 MB spi flash connected to Designware Subsystem SPI dirver, no XIP support, just for data storage.

Currently we just have some simple drivers for internal flash and bootspi controller in embarc_osp. However, as no deailed datasheet, these drivers are developed according to the limitted description of registers. so no gurantee on the quality.

For a real flash runner,  it needs to  add C language flash diver in openocd. It seems that using tcl script cannot implement the flash runner.  We need help from ARC openocd team.

vonhust, 2019-08-10:
Now, openocd in zephyr sdk is ok to load application into ram. Flash driver (internal eflash or outside spi flash) support will be implemented in the future. So I don't close this issue, but lower its priority

abrodkin, 2020-12-07:
@nashif if there's no demand on getting IoTDK's real flash usable for loading Zephyr I suggest to close this one so it no longer lingers here open.

closed 2021-09-25 by nashif