zephyrproject-rtos/zephyr #15814: [Coverity CID :186196]Unchecked return value in  samples/sensor/lsm6dsl/src/main.c

maxxlife, 2019-05-02:
Unchecked return value in File:  samples/sensor/lsm6dsl/src/main.c 
Category: Error handling issues
Function: sensor_trigger_set
Component: Samples
CID: 186196
Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

maxxlife, 2019-05-02:
Fixed by #15743

MaureenHelm, 2019-05-02:
> Fixed by #15743

If you put this link in the PR instead, the issue will automatically close when the PR gets merged. See https://help.github.com/en/articles/closing-issues-using-keywords

maxxlife, 2019-05-02:
> > Fixed by #15743
> 
> If you put this link in the PR instead, the issue will automatically close when the PR gets merged. See https://help.github.com/en/articles/closing-issues-using-keywords

Ok, I did it:)

ioannisg, 2019-07-04:
OK, but this is still open

ioannisg, 2019-07-04:
Fixed

closed 2019-07-04 by ioannisg