zephyrproject-rtos/zephyr #15817: nrf52: HFXO is not turned off as expected

tsvehagen, 2019-05-02:
**Describe the bug**
I want to enter a system on sleep mode to minimize power usage. When running, Bluetooth mesh is being used but before going to sleep with k_cpu_idle, I suspend or reset it with bt_mesh_reset or bt_mesh_suspend. The problem is that HFXO is still in use and so a low power state is not really possible.

When CONFIG_BT_CTLR_XTAL_ADVANCED=n, the HFXO is turned off as expected. m16src_ref is 1 and goes to 0 after mesh reset.

**To Reproduce**
Steps to reproduce the behavior:
1. Build Zephyr with CONFIG_BT_CTLR_XTAL_ADVANCED=y
2. Enable bt mesh using bt_enable, bt_mesh_init and bt_mesh_provision
3. m16src_ref in drivers/clock_control/nrf_power_clock.c is now 2, not sure if this is correct
3. Call bt_mesh_reset
4. Read NRF_CLOCK->HFCLKSTAT to see that HFXO is still on (0x10001)
5. Also m16src_ref is still 1 (ie one reference)

**Expected behavior**
In step 4 above I would expect HFXO to be off, HFCLKSTAT = 0x10000
In step 5 above I would expect m16src_ref to be 0

**Impact**
No way to enter a system on sleep mode with low power usage.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain Zephyr SDK 0.10.0
 - Zephyr v1.14.0

jhedberg, 2019-05-02:
The description doesn't mention any specific sample, but I suppose `tests/bluetooth/mesh_shell` might be most convenient using the following shell commands (didn't try myself but it should cause the same APIs to be called):
```
mesh init
mesh provision 1 2
mesh reset
```

closed 2019-05-03 by carlescufi

summer20100514, 2019-07-10:
@tsvehagen hi, after the patch do you get good power consumption result?

PR #15830 (2019-05-03): Bluetooth: controller: Fix XTAL advanced feature
Fix XTAL advanced feature by adding the missing
implementation to calculate and, retain or release the XTAL
clock source after a Bluetooth state or role is stopped.

Fixes #15817.
commit dbccf8e4 (2019-05-03): Bluetooth: controller: Fix XTAL advanced feature

Fix XTAL advanced feature by adding the missing
implementation to calculate and, retain or release the XTAL
clock source after a Bluetooth state or role is stopped.

Fixes #15817.
commit 2d1e079f (2019-05-03): Bluetooth: controller: Fix XTAL advanced feature

Fix XTAL advanced feature by adding the missing
implementation to calculate and, retain or release the XTAL
clock source after a Bluetooth state or role is stopped.

Fixes #15817.
commit 211a81f9 (2019-05-03): Bluetooth: controller: Fix XTAL advanced feature

Fix XTAL advanced feature by adding the missing
implementation to calculate and, retain or release the XTAL
clock source after a Bluetooth state or role is stopped.

Fixes #15817.