zephyrproject-rtos/zephyr #15823: Build failure for spi_loopback on atsamr21_xpro

galak, 2019-05-02:
https://app.shippable.com/github/zephyrproject-rtos/ci-test/runs/2909/3/tests

```

-- Configuring done
CMake Error at ../../../../cmake/extensions.cmake:378 (add_library):
  No SOURCES given to target: drivers__dma
Call Stack (most recent call first):
  ../../../../cmake/extensions.cmake:355 (zephyr_library_named)
  ../../../../drivers/dma/CMakeLists.txt:3 (zephyr_library)

```

galak, 2019-05-02:
@benpicco

benpicco, 2019-05-02:
That's because https://github.com/zephyrproject-rtos/zephyr/pull/15023 was merged but https://github.com/zephyrproject-rtos/zephyr/pull/14861 was not.

Sizurka, 2019-05-02:
This appears to because #15023 was merged without its dependency (#14861).  I have no clue why it didn't trip CI initially, however (maybe it didn't reload Kconfig for some reason?).

Basically #15023 added async SPI, but also added `select DMA if SPI_ASYNC` to SAM0 SPI, and without the actual DMA driver from #14861 there's no SAM0 DMA driver.

galak, 2019-05-02:
> That's because #15023 was merged but #14861 was not.

Thanks, just reviewed #14861.

galak, 2019-05-03:
Resolved with merge of #14861

closed 2019-05-03 by galak