zephyrproject-rtos/zephyr #15831: qemu_x86 DTS does not reflect actual hardware layout

andrewboie, 2019-05-02:
The QEMU x86 emulator has 8MB reserved on its command line.
In our QEMU DTS, this was cut in half; the first 4MB (minus the zero page) was reserved for a simulated XIP ROM area. The second 4MB is read-write system RAM.

When the flash simulator driver was introduced, the ROM area was reduced to 1MB. This breaks a bunch of tests when CONFIG_COVERAGE=y as they require more than 1MB.

We should not have done this. DTS is supposed to reflect the ACTUAL hardware configuration. If we have a software constructed fake driver, its characteristics should be in Kconfig, not by hacking the DTS files.

At this time I am unable to raise the ROM size back where it was. I tried to de-couple the simulated flash characteristics from DTS but there are overlays and flash partition layouts depending on it by other tests. The closest I got was #15828 

**Please put the qemu_x86 flash size back to its original value of 4092K**

andrewboie, 2019-05-02:
Is there a way we can get this flash simulator driver to operate on the actual RAM area in the emulator devoted to it?

In other words, can we get rid of the "static u8_t mock_flash[FLASH_SIZE]" and instead have the driver directly operate on CONFIG_FLASH_SIMULATOR_BASE_ADDRESS? We'll need test code not to blow away actual text/ROM, and adjust the MMU permissions of course.

lemrey, 2019-05-03:
Hi @andrewboie the driver wasn't meant, at least initially, to run on emulated targets only, so it needed an actual RAM buffer. I suppose that could be changed like you suggested if we don't need to run it on actual hardware.

andrewboie, 2019-05-03:
> Hi @andrewboie the driver wasn't meant, at least initially, to run on emulated targets only, 

That makes no sense to me, since the driver requires that the device have much more RAM and ROM since you decided to make the characteristics of the fake flash come from DTS...in the real world typically the reverse is true

kapi-no, 2019-05-06:
> That makes no sense to me, since the driver requires that the device have much more RAM and ROM since you decided to make the characteristics of the fake flash come from DTS...in the real world typically the reverse is true

Just for clarification:

The configuration was initially placed in the simulator's Kconfig by @lemrey. Then, I moved it to DTS as suggested by reviewers in this thread: https://github.com/zephyrproject-rtos/zephyr/pull/8254#discussion_r239204669

To make it all fit, I reduced the ROM area to 1MB - the CI test has passed for this DT configuration and I wasn't aware of any additional tests that fail.

> We should not have done this. DTS is supposed to reflect the ACTUAL hardware configuration.

I assumed (probably wrongly) that ROM value for QEMU DT can be modified since its only the simulator and I was further encouraged by the example of RAM configuration in the qemu_x86 dts file, which was dependent on the XIP setting (so configurable).

andrewboie, 2019-05-06:
> I assumed (probably wrongly) that ROM value for QEMU DT can be modified

The design of the flash simulator has a fundamental problem: it requires that the target platform have more RAM space than ROM space, because a buffer equal to the ROM size is allocated in the driver, in RAM. Not only that, but for its associated test case to pass, you need over twice as much RAM than ROM because the testcase itself also allocates a buffer equal in size to ROM.

This is not workable. We need a different solution that doesn't impose unreasonable constraints on the board DTS like this.

nvlsianpu, 2019-05-10:
@andrewboie 
> The QEMU x86 emulator has 8MB reserved on its command line.
In our QEMU DTS, this was cut in half; the first 4MB (minus the zero page) was reserved for a simulated XIP ROM area. The second 4MB is read-write system RAM.

Is that possible to extend  this 8MB of memory to 12MB? So we can use 8MB for read-write system RAM so even if enabled flash-simulator we still keep free 4 MB? I'm not familiar with qemu usage well so mayby ther is a better solution for the issue?

You state is that we should move at last flash-simulator NVM size to Kconfig as it was originally proposed?

nvlsianpu, 2019-05-10:
@andrewboie #16070

andrewboie, 2019-06-04:
This is still broken, ROM size is still 1MB:
```

[ 92%] Linking C executable zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
             ROM:     1406720 B         1 MB    134.16%
             RAM:       1464 KB         4 MB     35.74%
        IDT_LIST:         464 B         2 KB     22.66%
        MMU_LIST:         120 B         1 KB     11.72/opt/zephyr-sdk/i586-zephyr-elf/bin/../lib/gcc/i586-zephyr-elf/8.3.0/../../../../i586-zephyr-elf/bin/ld: zephyr_prebuilt.elf section `text' will not fit in region `ROM'

```

carlescufi, 2019-06-06:
@nvlsianpu can you please take a look?

nvlsianpu, 2019-06-06:
will do, tomorrow.

ioannisg, 2019-06-26:
@nvlsianpu ping :)

nvlsianpu, 2019-07-03:
To solve this issue need to solve https://github.com/zephyrproject-rtos/zephyr/issues/16773 before. I'm not able to do so.

nvlsianpu, 2019-07-04:
hopefully fix in a way by #17349

nvlsianpu, 2019-07-29:
@andrewboie I believe this is already fixed - can you confirm and close the issue?

closed 2019-07-29 by andrewboie