zephyrproject-rtos/zephyr #15835: "#if XIP" block in qemu_x86 DTS always evaluates to false

andrewboie, 2019-05-02:
The intent here seems to be to properly set memory if CONFIG_XIP=y, but the qemu_x86 target does enable CONFIG_XIP, yet the amount of SRAM gets set to 8188K anyway.

Causes hard to debug runtime hangs if the amount of RAM used exceeds 4096, instead of failing at build time.

galak, 2019-05-03:
Not sure how this snuck in, but we don't allow use of CONFIG_ ifdef's in dts anymore.

nashif, 2019-05-03:
This is not a CONFIG_* and it was there for almost 2 years, without any effect basically. We do not define XIP anywhere AFAIK

```
#if XIP
        #define DT_SRAM_SIZE            DT_SIZE_K(4096)
#else
        #define DT_SRAM_SIZE            DT_SIZE_K(8188)
#endif
```

PR #15894 (2019-05-04): boards: qemu_x86: properly enable XIP
The '#if XIP' in the DTS file never worked properly,
causing the QEMU build to think it has much more RAM
then it actually has. If RAM overflowed, this would not
be caught by the build, instead there would be strange
crashes when the data copy takes place.

The QEMU targets themselves are not XIP, everything
is actually RAM, but the first 4 megabytes are
considered to be a memory-mapped flash region. This
is done to ensure that the XIP data copying infrastructure
doesn't bit-rot on x86. We are at the point where
a lot of things depend on this, so just sele [... cut]
commit 87d1f068 (2019-05-04): boards: qemu_x86: properly enable XIP

The '#if XIP' in the DTS file never worked properly,
causing the QEMU build to think it has much more RAM
then it actually has. If RAM overflowed, this would not
be caught by the build, instead there would be strange
crashes when the data copy takes place.

The QEMU targets themselves are not XIP, everything
is actually RAM, but the first 4 megabytes are
considered to be a memory-mapped flash region. This
is done to ensure that the XIP data copying infrastructure
doesn't bit-rot on x86. We are at the point where
a lot of things depend on this, so just sel [... cut]
closed 2019-05-04 by nashif