zephyrproject-rtos/zephyr #15841: Support AT86RF233

KwonTae-young, 2019-05-03:
**Is your feature request related to a problem? Please describe.**
I'm trying to use [AT86RF233](https://www.microchip.com/wwwproducts/en/AT86RF233) for wireless communication.
My first goal is a 2.4GHz RF wireless keyboard.
SAM R21 was supported in #13494, but currently [AT86RF233](https://www.microchip.com/wwwproducts/en/AT86RF233) support is missing.

**Describe the solution you'd like**
I want a driver to support [AT86RF233](https://www.microchip.com/wwwproducts/en/AT86RF233).
I want to operate the wireless keyboard with 2.4GHz communication(2Mbps) of [AT86RF233](https://www.microchip.com/wwwproducts/en/AT86RF233).

**Describe alternatives you've considered**
I have tested two [ESB(Enhanced Shock Burst)](https://devzone.nordicsemi.com/b/blog/posts/intro-to-shockburstenhanced-shockburst) communications with two [nRF52840-PCA10056](https://docs.zephyrproject.org/latest/boards/arm/nrf52840_pca10056/doc/index.html) boards.(#15607)
test video: https://youtu.be/8rfX-1xxlSI
But it must use nordic MCU.

**Additional context**
I have no experience dealing with this type of wireless chip.
Sources that use Arduino and [AT86RF233](https://www.microchip.com/wwwproducts/en/AT86RF233) exist in other sites.([arduino-at86rf233](https://github.com/msolters/arduino-at86rf233))

I connected [ST STM32F4DISCOVERY](https://docs.zephyrproject.org/latest/boards/arm/stm32f4_disco/doc/index.html) and [REB233-XPRO](https://www.microchip.com/developmenttools/ProductDetails/atreb233-xpro) as follows.
The [REB233-XPRO](https://www.microchip.com/developmenttools/ProductDetails/atreb233-xpro) is an AT86RF233 module.
[image]

pin | AT86RF233(REB X-Pro Extension Header) | STM32F4-DISCO |
---|---|---
Reset | 7 | PB0
Interrupt | 9 | PB1
Sleep | 10 | PB2
CS | 15 | PA4
MOSI | 16 | PA7
MISO | 17 | PA6
SCK | 18 | PA5
Ground | 19 | GND
Power | 20 | VDD (3V)

[image]

And I read the part number, version, and main id through SPI communication.

```
west init zephyrproject -m https://github.com/KwonTae-young/zephyr/ --mr at86rf233
cd zephyrproject/zephyr
source zephyr-env.sh
mkdir samples/drivers/rf/spi_at86rf2xx/build
cd samples/drivers/rf/spi_at86rf2xx/build
cmake -DBOARD=stm32f4_disco ..
make -j8 flash
```

**console output**
```
***** Booting Zephyr OS zephyr-v1.13.0-6846-gf3409e84cf63 *****
AT86RF2XX(2.4GHz Transceiver) example application

PART_NUM:0xb
VERSION_NUM:0x1
MAIN_ID:0x1f
```

I would like help from people who are interested in [AT86RF233](https://www.microchip.com/wwwproducts/en/AT86RF233).

tbursztyka, 2019-05-03:
> My first goal is a 2.4GHz RF wireless keyboard.

Do you know the protocol used by the keyboard?
Zephyr provides support for IEEE 802.15.4 + 6LoWPAN, but nothing for Zigbee (which is kind of proprietary I believe. Correct me if I am wrong). 

Since it is an Atmel chip, ASF should provide support for it. It would require to import the necessary bits from ASF into ext/hal/atmel/asf.

If and only if you are interested in 802.15.4: 
It is "just" a matter of implementing a zephyr driver using this HAL and implementing this interface include/net/ieee802154_radio.h , see drivers/ieee802154/ for some examples

KwonTae-young, 2019-05-03:
@tbursztyka 
> Do you know the protocol used by the keyboard?

I want to wirelessly receive the data and output it to the `HID keyboard`.
I have tried HID keyboard.

<br/>

> It would require to import the necessary bits from ASF into ext/hal/atmel/asf.

Later, I'm trying to reference the ASF.
But I am still immature at the use of Atmel.:sweat_smile:
<br/>

> If and only if you are interested in 802.15.4

I have not planned to use IEEE 802.15.4 yet.
In some cases, you can send a string, but IEEE 802.15.4 has a slower bit rate (250kbps).

I'm sorry I do not have enough English. :)
Thank you.

closed 2019-09-19 by KwonTae-young

nandojve, 2020-02-08:
Hi @KwonTae-young, if you are interested yet on RF2xx driver please take a look at #20296. We merged a full functional 2.4GHz driver. It was tested by Tridonic for Openthread compatibility too.
I hope it can help you!

p0we7, 2021-08-16:
Hello, @KwonTae-young , from your description it seems that you have completed a prototype of the NRF52840 using ESB in Zephyr, can you provide more information on this?

KwonTae-young, 2021-08-17:
> Hello, @KwonTae-young , from your description it seems that you have completed a prototype of the NRF52840 using ESB in Zephyr, can you provide more information on this?

@p0we7 ESB-related examples were located in other repositories.(#15607)  
https://github.com/NordicPlayground/fw-nrfconnect-nrf/tree/master/samples/esb