zephyrproject-rtos/zephyr #15842: cdc_acm: stm32: `uart_fifo_fill` can't transmit data out

jli157, 2019-05-03:
**Describe the bug**
A clear and concise description of what the bug is.
Running the sample `samples/subsys/usb/cdc_acm` on the board `nucleo_f429zi` to receive data written by a host program and echo back to the host. However, Echo doesn't work. The board can only receive data but can't send data out. 

**To Reproduce**
Steps to reproduce the behavior:
```bash
$ cd samples/subsys/usb/cdc_acm
$west build -b nucleo_f429zi
$west flash
```
Connect the board's SoC USB to a computer, and run the attached python script:
  ```bash
   $python3 test_uart.py /dev/ttyACM1
```
5. the test script sends one text line to the board every 1 second, and expects to receive it from the board. 

**Expected behavior**
The test_uart.py should receive the echoed strings from the board. 

**Impact**
TX on CDC_ACM doesn't work. 

**Screenshots or console output**

- Host side
```
 $python3 test_uart.py /dev/ttyACM1 
Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!

Write: This is a stm32 F429zi packet!
```

Device side:
```
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
This is a stm32 F429zi packet!
```

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04 in VM
 - Toolchain: Zephyr SDK-0.10.0
 - Commit SHA or Version used: 54e78b9 on master branch 
- Modified `cdc_acm/src/main.c` to output received strings:
```
diff --git a/samples/subsys/usb/cdc_acm/src/main.c b/samples/subsys/usb/cdc_acm/src/main.c
index f4d50a2483..0a25ade5bb 100644
--- a/samples/subsys/usb/cdc_acm/src/main.c
+++ b/samples/subsys/usb/cdc_acm/src/main.c
@@ -68,6 +68,7 @@ static void read_and_echo_data(struct device *dev, int *bytes_read)
        /* Read all data and echo it back */
        while ((*bytes_read = uart_fifo_read(dev,
            (u8_t *)data_buf, sizeof(data_buf)))) {
+               printk("%s", data_buf);
                write_data(dev, data_buf, *bytes_read);
        }
 }
```

**Additional context**
Add any other context about the problem here.

jli157, 2019-05-03:
## Test script attached here:

[test_uart.py.txt](https://github.com/zephyrproject-rtos/zephyr/files/3140850/test_uart.py.txt)

jli157, 2019-05-03:
@finikorg Please help to check this issue. It is really a blocking issue for my project.

jli157, 2019-05-03:
Tried with a terminal on the host side, and got the same result.

jli157, 2019-05-03:
@finikorg Verified the same sample app on `zephyr-v1.14.0`. Everything works well. 

```
Write: This is a stm32 F429zi packet!

rcv: b'This is a stm32 F429zi packet!\r\n'
Write: This is a stm32 F429zi packet!

rcv: b'This is a stm32 F429zi packet!\r\n'
```

finikorg, 2019-05-03:
Does PR #15755 fixes the issue?

jli157, 2019-05-03:
> Does PR #15755 fixes the issue?

Yes, TX works after applying this patch. Thank you!

PR #15755 (2019-05-06): USB: Fix endpoint allocation order 
Fix allocation order for endpoints and add tests illustrating the issue

Fixes #15842
closed 2019-05-06 by nashif