zephyrproject-rtos/zephyr #15844: Network management API should support user space

jukkar, 2019-05-03:
The net_mgmt API should be usable from user space application as it might want to listen to events that the network stack is sending.

tbursztyka, 2019-05-03:
We have a problem here: net_mgmt was designed for tiny non-mmu based system so it's using callbacks for events.

We might have to rethink its design or at least the way it throws events to user space.

jukkar, 2019-05-03:
Yes, the callback system will not work very well with user space. We could for example place fifo between kernel and user space and transfer the event information via it.

andrewboie, 2019-05-03:
The current expectation is that the application will register their callback functions, which will then use some sort of IPC mechanism to wake up a thread to do further processing. This is more or less an exercise for the end user, although there is a dearth of samples at the moment to show the best way to do this (this is being worked on in #14683 for 1.15)

It will probably be of some value to introduce a generic subsystem, not specific to networking, to make it easier to indirectly attach user threads or workqueues to callbacks (networking, bluetooth, drivers, timers, etc) without writing a lot of boilerplate. At the very base level, I think there will still be callbacks involved that run in interrupt context.

What I'm getting at here is that there is a need for this, but this would be built on top of existing APIs, and not necessarily specific to networking.

rerickson1, 2019-06-12:
I am using zephyr 1.14 and am trying to use `net_mgmt_event_callback` in one of my app threads.  I do not see the callback firing properly when the iface goes up and down.  Am I not allowed to use the callbacks in my app?

jukkar, 2019-07-03:
Fixed by #16743 which is already merged -> closing

closed 2019-07-03 by jukkar