zephyrproject-rtos/zephyr #15864: disk partitioning should not specified in DTS

andrewboie, 2019-05-03:
flash partitioning is a software construction. it doesn't have anything to do with the physical hardware (although it depends on the size of that hardware). why is the flash layout  expressed in DTS and not in code or somewhere else?

please find some other way to specify software constructions like flash partitions. DTS should be used to specify immutable hardware characteristics only

galak, 2019-05-03:
There are some software constructs that are described in DTS, and the reason to have the flash partition in DTS is it provides a portable way to describe this information across different software projects.

andrewboie, 2019-05-03:
> There are some software constructs that are described in DTS, and the reason to have the flash partition in DTS is it provides a portable way to describe this information across different software projects.

Where do you draw the line?
Why not get rid of Kconfig and put all of that in DTS as well?

galak, 2019-05-03:
I think its reasonable to take it on a case by case basis.  If there truly is something that reasonable needs to be know between software components than its reasonable to see if DTS is a way to handle that.  DTS already had a partition definition and thus we used it.

andrewboie, 2019-05-03:
> I think its reasonable to take it on a case by case basis. If there truly is something that reasonable needs to be know between software components than its reasonable to see if DTS is a way to handle that. DTS already had a partition definition and thus we used it.

I'm worried about scope creep, but I'll give in on this.

closed 2019-05-03 by andrewboie