zephyrproject-rtos/zephyr #15881: tests/net/buf fails on qemu_x86_64

andrewboie, 2019-05-03:
```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..Running test suite net_buf_test
===================================================================
starting test - net_buf_test_1
PASS - net_buf_test_1
===================================================================
starting test - net_buf_test_2
PASS - net_buf_test_2
===================================================================
starting test - net_buf_test_3

    Assertion failed at ./tests/net/buf/src/main.c:195: net_buf_test_3: (k_sem_take(&sema, TEST_TIMEOUT) == 0 is false)
Timeout while waiting for semaphore
FAIL - net_buf_test_3
===================================================================
starting test - net_buf_test_4

    Assertion failed at ./tests/net/buf/src/main.c:156: test_3_thread: (buf is NULL)
Unable to get buffer

```

jukkar, 2019-05-03:
@jhedberg can you check this one? I looked this briefly but did not quite got why the test was failing.

andrewboie, 2019-05-03:
It's quite plausible that this could ultimately be an x86_64 issue, in which case we should bring to @andyross 's attention

jukkar, 2019-08-07:
I remember checking net_buf tests briefly earlier but my notes seem to have lost somewhere. After adding some printk()'s to the code, it seems that the semaphores in net_buf_test_3() are not waited and they fail immediately regardless of the timeout value TEST_TIMEOUT. To me it looks the issue is in qemu_x86_64 and not in the net_buf unit test.

andyross, 2019-08-20:
Swung back by to check this: this passes fine with the fixes in #18531, though it requires that we remove the whitelist so it actuallys runs.

nashif, 2019-08-29:
this platform is not whitelisted for this list and per @andyross the test passes with #18531 if platform is enabled, so lowering priority as there is no impact to the 2.0 release.

nashif, 2019-11-20:
can't reproduce

closed 2019-11-20 by nashif