zephyrproject-rtos/zephyr #15884: tests/net/socket/getaddrinfo fails on mps2_an385

andrewboie, 2019-05-03:
```
qemu-system-arm: warning: nic lan9118.0 has no peer
***** Booting Zephyr OS zephyr-v1.14.0-448-g869a1d7b5161 *****
Running test suite socket_getaddrinfo
===================================================================
starting test - test_getaddrinfo_setup
PASS - test_getaddrinfo_setup
===================================================================
starting test - test_getaddrinfo_ok
***** MPU FAULT *****
  Data Access Violation
  MMFAR Address: 0x20020000
***** Hardware exception *****
Current thread ID = 0x200010cc
Faulting instruction address = 0x21e
Fatal fault in thread 0x200010cc! Aborting.
QEMU: Terminated
[100%] Built target debugserver
```

Seems to be a problem with newlib heap configuration wrt user mode, I'll look into this one.

ioannisg, 2019-06-30:
This one does not fail any more, @andrewboie. Could this have been fixed by all ARM- or socket-related bugfixes?

ioannisg, 2019-06-30:
```
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m3
qemu-system-arm: warning: nic lan9118.0 has no peer
***** Booting Zephyr OS build zephyr-v1.14.0-1930-gf7697c43b934 *****
Running test suite socket_getaddrinfo
===================================================================
starting test - test_getaddrinfo_setup
PASS - test_getaddrinfo_setup
===================================================================
starting test - test_getaddrinfo_ok
PASS - test_getaddrinfo_ok
===================================================================
starting test - test_getaddrinfo_cancelled
PASS - test_getaddrinfo_cancelled
===================================================================
starting test - test_getaddrinfo_no_host
PASS - test_getaddrinfo_no_host
===================================================================
starting test - test_getaddrinfo_num_ipv4
PASS - test_getaddrinfo_num_ipv4
===================================================================
starting test - test_getaddrinfo_flags_numerichost
PASS - test_getaddrinfo_flags_numerichost
===================================================================
Test suite socket_getaddrinfo succeeded
===================================================================
PROJECT EXECUTION SUCCESSFUL

```

andrewboie, 2019-06-30:
I fixed this, we can close it.

closed 2019-06-30 by andrewboie