zephyrproject-rtos/zephyr #15893: code coverage is not tested in CI

andrewboie, 2019-05-03:
To enable code coverage, the special Kconfig option CONFIG_COVERAGE must be enabled.

I have found that many tests are failing when this is turned on. Most of this is due to stack overflows, but I have found a variety of issues. We need to stop this from bit-rotting, ensuring code coverage is vital for functional safety requirements.

I am going to introduce qemu_x86_coverage and mps2_an385_coverage targets. These have the same configuration but enabled CONFIG_COVERAGE=y. We will then show that coverage is working properly. The first patch is #15886 

We also need to put a stop to a trend to whitelist test targets without good reason. #15836 PR does this for network test cases, but there are others.

aescolar, 2019-05-04:
Related sidetrack: Ideally we would also collect and track coverage. I know the codecov solution was not optimal, and its results were confusion people sometimes, but it would be nice to have something like we did.

nashif, 2019-06-11:
configuration to test coverage was merged.

closed 2019-06-11 by nashif