zephyrproject-rtos/zephyr #15906: WEST ERROR: extension command build was improperly defined

maxxlife, 2019-05-04:
I'm trying to run west, but it seems that it is quite difficult to do.
I'm trying to repeat everythin what is written here https://docs.zephyrproject.org/latest/getting_started/index.html

That error I receive when I'm running that command `west build -b reel_board samples/hello_world`
`ZEPHYR_BASE=/home/maxxlife/zephyrproject/zephyr (origin: env)`

`ERROR: extension command build was improperly defined and could not be run: could not import` `/home/maxxlife/zephyrproject/zephyr/scripts/west_commands/build.py`

`Traceback (most recent call last):`
  `File "/home/maxxlife/zephyrproject/.west/west/src/west/commands/command.py", line 317, in __call__`
  `  mod = _commands_module_from_file(self.py_file)`
  `File "/home/maxxlife/zephyrproject/.west/west/src/west/commands/command.py", line 284, in _commands_module_from_file`
 `   mod = _import_mod_from(mod_name, file)`
 ` File "/home/maxxlife/zephyrproject/.west/west/src/west/commands/command.py", line 281, in _import_mod_from`
  `  spec.loader.exec_module(mod)`
 ` File "<frozen importlib._bootstrap_external>", line 678, in exec_module`
 ` File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed`
 ` File "/home/maxxlife/zephyrproject/zephyr/scripts/west_commands/build.py", line 12, in <module>
   ` from cmake import DEFAULT_CMAKE_GENERATOR, run_cmake, run_build, CMakeCache`
`ImportError: cannot import name 'DEFAULT_CMAKE_GENERATOR'`

`The above exception was the direct cause of the following exception:`

`Traceback (most recent call last):`
  File "/home/maxxlife/zephyrproject/.west/west/src/west/main.py", line 580, in main`
    args.handler(args, unknown)`
  File "/home/maxxlife/zephyrproject/.west/west/src/west/main.py", line 323, in ext_command_handler`
    command = spec.factory()`
 ` File "/home/maxxlife/zephyrproject/.west/west/src/west/commands/command.py", line 320, in __call__`
    `hint='could not import {}'.format(self.py_file)) from ie`
`west.commands.command.ExtensionCommandError`
Do you know any suggestions how to solve it?

maxxlife, 2019-05-04:
Just I need to be more careful^_^

closed 2019-05-04 by maxxlife

cito-lito, 2019-10-30:
im facing the same issue, how did you fixed it?

aeremin, 2020-02-24:
For the next poor soul who will google this error and will find this issue:
You probably missed this step:

## Install needed Python packages ##
`pip3 install --user -r ~/zephyrproject/zephyr/scripts/requirements.txt`