zephyrproject-rtos/zephyr #15916: [BLE] Mesh example qemu kernel panic

n3rd4i, 2019-05-05:
**Describe the bug**
A clear and concise description of what the bug is.
- Bluetooth Mesh example creates kernel panic with QEMU and Host Bluetooth dongle.

What have you tried to diagnose or workaround this issue? Yes

**To Reproduce**
Steps to reproduce the behavior:
1. cp -r $ZEPHYR_BASE/samples/bluetooth/mesh/ .
2. west build -b qemu_x86 .
3. west build -t run
4. See error:
```sh
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.14.0-511-g851fa12d51a4 *****
Initializing...
***** Kernel Panic! *****
Current thread ID = 0x005018a0
eax: 0xffffffea, ebx: 0x00000000, ecx: 0x00507120, edx: 0x00507134
esi: 0x0000033b, edi: 0x00000f50, ebp: 0x00505330, esp: 0x00505320
eflags: 0x00000282 cs: 0x0008
call trace:
eip: 0x00005c61
     0x00005c9e (0x505344)
     0x00009087 (0x5072ac)
     0x0000b22f (0x0)
     0x000003b9 (0x33b)
     0x00021cfd (0x50537c)
     0x00000f5f (0x0)
Terminate emulator due to fatal kernel error
FAILED: zephyr/CMakeFiles/run
cd /home/vagrant/mesh/build && /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 8 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -no-acpi -net none -pidfile qemu.pid -serial mon:stdio -serial unix:/tmp/bt-server-bredr -s -kernel /home/vagrant/mesh/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/bin/cmake --build /home/vagrant/mesh/build --target run
run as "west -v build -t run" for a stack trace
```

* meanwhile command: `sudo bluez/tools/btproxy -u -i 0`
```
Opening user channel for hci0
New client connected
Error from host descriptor
Closing host descriptor
Closing device descriptor
```

**Expected behavior**
Bluetooth Mesh example using QEMU with host BLE should work.

**Impact**
showstopper

**Environment (please complete the following information):**
 - OS: Linux (VM from Windows)
 - Toolchain: Zephyr SDK,
 - Commit: latest `west update` version

jhedberg, 2019-05-06:
Wouldn’t surprise me if this is a simple stack overflow. Could you try increasing the main thread, system work queue thread and Bluetooth RX thread stack sizes to see if the issue goes away?

n3rd4i, 2019-05-07:
I've doubled the values but I still get the errors:
```log
2,3c2,3
< CONFIG_MAIN_STACK_SIZE=512
< CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
---
> CONFIG_MAIN_STACK_SIZE=1024
> CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
29c29
< CONFIG_BT_RX_BUF_COUNT=30
---
> CONFIG_BT_RX_BUF_COUNT=60
```

jhedberg, 2019-05-08:
Could you also try increasing the RX stack size? The Kconfig option for it is `CONFIG_BT_RX_STACK_SIZE` (the `CONFIG_BT_RX_BUF_COUNT` you modified is just the number of data buffers for incoming HCI events & ACL data)

n3rd4i, 2019-05-08:
The new changes are as follows:
```log
[vagrant@archlinux build]$ diff $ZEPHYR_BASE/samples/bluetooth/mesh/prj.conf ../prj.conf 
2,3c2,4
< CONFIG_MAIN_STACK_SIZE=512
< CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=2048
---
> CONFIG_MAIN_STACK_SIZE=1024
> CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=4096
> CONFIG_BT_RX_STACK_SIZE=2048
```

Console output:
```log
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.14.0-605-ge9ed63465426 *****
Initializing...
***** Kernel Panic! *****
Current thread ID = 0x005018a0
eax: 0xffffffea, ebx: 0x00000000, ecx: 0x00507120, edx: 0x00507134
esi: 0x0000033b, edi: 0x00000f50, ebp: 0x00505530, esp: 0x00505520
eflags: 0x00000282 cs: 0x0008
call trace:
eip: 0x00005c61
     0x00005c9e (0x505544)
     0x00009087 (0x5072ac)
     0x0000b22f (0x0)
     0x000003b9 (0x33b)
     0x00021d36 (0x50557c)
     0x00000f5f (0x0)
Terminate emulator due to fatal kernel error
Error from host descriptor
Closing host descriptor
Closing device descriptor
FAILED: zephyr/CMakeFiles/run
cd /home/vagrant/projects/mesh/build && /opt/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/qemu-system-i386 -m 8 -cpu qemu32,+nx,+pae -device isa-debug-exit,iobase=0xf4,iosize=0x04 -no-reboot -nographic -no-acpi -net none -pidfile qemu.pid -serial mon:stdio -serial unix:/tmp/bt-server-bredr -s -kernel /home/vagrant/projects/mesh/build/zephyr/zephyr.elf
ninja: build stopped: subcommand failed.

```

jhedberg, 2019-05-23:
Could this be the same as #16275 ? @n3rd4i could you try disabling `CONFIG_BT_SETTINGS` in the configuration?

n3rd4i, 2019-05-27:
After setting: `CONFIG_BT_SETTINGS=n`
Things start working:
```
SeaBIOS (version rel-1.12.0-0-ga698c8995f-prebuilt.qemu.org)
Booting from ROM..***** Booting Zephyr OS zephyr-v1.14.0-605-ge9ed63465426 *****
Initializing...
Bluetooth initialized
Mesh initialized
[00:00:00.330,000] <inf> bt_hci_core: Identity: [...]
```

jhedberg, 2019-05-27:
> After setting `CONFIG_BT_SETTINGS=n` things start working

Ok, then once #16405 gets merged there should be a proper fix for this (no need to disable BT_SETTINGS). I'll close this issue now since it's a duplicate of an existing one.

closed 2019-05-27 by jhedberg