zephyrproject-rtos/zephyr #15917: USB disconnect/reconnect

acsplayon, 2019-05-05:
Hardware/Software
nrf52840 on Nordic PCA100056 DK
USB configured for either CDC-ECM or CDC_ACM

** Describe the bug **
Detaching the USB cable from the host and reattaching (not timing dependent) will not reinitialize/reenumerate the device correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. build the cdc_acm sample for your board
2. flash and run on hardware
3. verify that USB device enumerates property on initial attach (or attached at boot)
4. pull USB cable from host, reattach after some period of time
5. verify that the USB device no longer words or enumerates on the host.

**Expected behavior**
The USB device should re-enumerate correctly and become functional after the reconnect.

**Impact**
Stability of product.  This would affect almost all users of USB.

**Screenshots or console output**

**Environment (please complete the following information):**
 - OS: Linux (ubuntu 16.04 and/or 18.04
 - Toolchain Zephyr SDK, 
 

**Additional context**
I have made a few changes to the code to provide a work-around/fix, however I believe the biggest issue is the architecture of the stack. usb structures do not get properly shutdown or disabled upon usb detach.
The first part of my fix, is to clear the context enabled flag for each end-point in ep_ctx_reset. It was found that upon reenumeration, the function usb_dc_ep_enable was unable to enable the endpoint because this flag was set. This function then returned an error code that was unexpected by the caller, which made the caller fail.
Fix 1 allows the usb device to reenumerate correctly, however, transaction records are lost when this happens (because transactions are pending at time of reconnect).  Eventually, enough transactions are lost that the usb_device code can no longer start transactions and reenumeration will again start failing (multiple detach/reattach attempts must be made).
Fix part 2. This fix includes a loop that will cancel all pending transactions on USB_DC_RESET messages.  This allows the kernel to free the semaphores that have been locked on pending transactions.  This also works, but then enters another issue.
Issue 3.  During the USB_DC_RESET sequence, higher level parts of the stack have not been informed of the condition of the USB bus, and try to continue sending/receiving USB transactions over interface endpoints (non-control endpoints).  These new transaction requests can come in at any time, even after cancelling transactions.  Again, these transactions are lost by the system, and semaphores remain locked, eventually allowing the usb_device to run out of usb transaction slots and incapable of starting new transactions.
Fix 3.  A new BOOL has been added to the usb_dev_priv local structure that gets cleared on USB_DC_RESET event, and set only after a USB_DC_INTERFACE event.  The function usb_transfer() has been modified to check this flag before starting a new transaction and if the new transaction is destined for any endpoint other than the control endpoint ((ep & 0x7F)==0) the function will return an error to the caller and the transaction not started.

These 3 fixes now make my devices and hardware stable on USB disconnect and reconnects. The total changes amount to about 15 lines of code in usb_device.c and usb_dc_nrfx.c.

masz-nordic, 2019-05-07:
I was unable to reproduce, at 859c4ed2cd41aa0ef5b5a29b3d2c64199b252ee1 with Ubuntu 16.04.

jfischer-phytec-iot, 2019-05-07:
@acsplayon As you already have the patch, could you please open a PR?

carlescufi, 2019-05-14:
@acsplayon what version of Zephyr are you testing with? There was a similar bug but was fixed a while back. Also, as @jfischer-phytec-iot mentioned, can you please post the code you have?

masz-nordic, 2019-05-15:
After revisiting: SetConfiguration(1) stalls on reattach without chip restart.
Bisection leads to 2b58594e9033eda383534270b15237a51df3f485
@carlescufi I think this issue should have higher priority as it breaks USB3CV tests too.

finikorg, 2019-05-15:
> After revisiting: SetConfiguration(1) stalls on reattach without chip restart.
> Bisection leads to [2b58594](https://github.com/zephyrproject-rtos/zephyr/commit/2b58594e9033eda383534270b15237a51df3f485)
> @carlescufi I think this issue should have higher priority as it breaks USB3CV tests too.

Wasn't the issue fixed in 

```
commit 714c8ff28ef2655d14457528618c92bdbe2094b9
Author: Johann Fischer <j.fischer@phytec.de>
Date:   Sat Apr 27 01:03:38 2019 +0200

    usb: usb_device: fix set configuration request
    
    Fix set configuration request for a configuration without
    specific endpoint like USB DFU class.
    
    The changes introduced in commit 2b58594e9033
    ("usb: device: Use set_endpoint helper for set_config")
    does not take into account that a configuration could only
    contain control endpoint.
```

finikorg, 2019-05-15:
I will check this issue

finikorg, 2019-05-15:
Apparently we fail because we have added extra checks, in this case endpoints are already enabled and re-enabling them fails. Right way would be disabling endpoints on cable disconnect.

carlescufi, 2019-05-15:
@finikorg I know that @barsok is looking at this, can you provide a PR or code snippet?

acsplayon, 2019-05-15:
@finikorg / Andrei.  I am sorry that I don't have time to do a pull request and update my code, as with most  engineers always under the gun.  You will find that just disabling the endpoints upon detach will not be enough to completely correct the bug (it is only 1 part of 3).  If you read the original bug request I document what I found in my investigation of this issue. Fixing the endpoints will allow the device to reconfigure, but multiple disconnects will cause semaphores to remain locked and eventually USB will fail again.
I have been writing software longer than many of you probably have been alive (I am not trying to insult) and am old school and have never learned the ins and outs of git.  I would be happy to send along my fixes because that takes little time, but learning the correct way to use git without breaking something would consume too much of my time.  I could certainly send an email with the source code changes. 
Also, I am testing on the master branch, and the previous bug listed about reconfiguration has nothing to do with this bug.

carlescufi, 2019-05-15:
@acsplayon thanks for the feedback and help fixing this. Can you simply do something like:

`git show <commit SHA>`
if you have committed your fixes in a particular commit? Then you can paste that output in a GitHub Gist:

https://gist.github.com/

or directly in a comment in this issue.

carlescufi, 2019-05-15:
Also feel free to join our Slack channel: https://tinyurl.com/yarkuemx and we can help you out there

acsplayon, 2019-05-15:
I have created a gist with the output of 'git diff <file>' on the two files I needed to modify.  I didn't not revisit all of the different drivers for all of the supported USB chips, only the Nordic driver because that is the hardware I have.

https://gist.github.com/acsplayon/b594d6b2207f8923edb071fdee499b36

carlescufi, 2019-05-15:
> I have created a gist with the output of 'git diff ' on the two files I needed to modify. I didn't not revisit all of the different drivers for all of the supported USB chips, only the Nordic driver because that is the hardware I have.
> 
> https://gist.github.com/acsplayon/b594d6b2207f8923edb071fdee499b36

Thank you @acsplayon, much appreciated!

@barsok @finikorg can one of you take a look at the patch in the file and submit a PR as appropriate?

jfischer-phytec-iot, 2019-05-15:
@loicpoulain please take a look, particularly at the re-initialization of the transfer slots after DC_RESET event.

loicpoulain, 2019-05-15:
Ok I'm going to have à look and test this change.

finikorg, 2019-05-16:
I am thinking about following approach

```
@@ -941,6 +944,12 @@ static void forward_status_cb(enum usb_dc_status_code status, const u8_t *param)
 {
        size_t size = (__usb_data_end - __usb_data_start);
 
+       if (status == USB_DC_DISCONNECTED) {
+               LOG_DBG("Disconnect");
+
+               usb_cancel_transfers();
+       }
+
        for (size_t i = 0; i < size; i++) {
                struct usb_cfg_data *cfg = &__usb_data_start[i];
 
@@ -1297,7 +1306,9 @@ done:
                k_sem_give(&trans->sem);
 
                /* Transfer completion callback */
-               cb(ep, tsize, priv);
+               if (trans->status != -ECANCELED) {
+                       cb(ep, tsize, priv);
+               }
        }
 }
 
@@ -1419,6 +1430,24 @@ done:
        irq_unlock(key);
 }
 
+void usb_cancel_transfers(void)
+{
+       for (int i = 0; i < ARRAY_SIZE(usb_dev.transfer); i++) {
+               struct usb_transfer_data *trans = &usb_dev.transfer[i];
+               unsigned int key;
+
+               key = irq_lock();
+
+               if (trans->status == -EBUSY) {
+                       trans->status = -ECANCELED;
+                       k_work_submit(&trans->work);
+                       LOG_DBG("Cancel transfer");
+               }
+
+               irq_unlock(key);
+       }
+}
+
 struct usb_transfer_sync_priv {
        int tsize;
        struct k_sem sem;
```

PR #16193 (2019-05-22): USB: Fix reconnect issues
I still need to check suspend / resume properly working.

Fixes #15917
closed 2019-05-22 by carlescufi