zephyrproject-rtos/zephyr #15922: BLE mesh:The Provisioner APP can't find the micro:bit which is running the mesh sample

jerryxiajie, 2019-05-06:
**Describe the bug**
I test samples/bluetooth/mesh with the Micro:bit and a nRF Mesh IOS APP.It seems that the micro:bit works well because the led matrix is blinking and the serial console shows the mesh node has been initialized. But the app can't discover any mesh node. I also tried the Silicon Labs BluetoothMesh APP but it still can't discover the micro:bit mesh node.

**To Reproduce**
Steps to reproduce the behavior:
1. cd zephyr/samples/bluetooth/mesh/
2. mkdir build; cd build
3. cmake -GNinja -DBOARD=bbc_microbit ..
4. ninja
5. cp build/zephyr/zephyr.hex to Micro:bit
6. Scan the mesh node on nRF Mesh App/Silicon Labs BluetoothMesh APP

**Expected behavior**
I can find the micro:bit mesh node and provision it through the provisioner APP.

**Screenshots or console output**
This is the logs from serial cosole:
`***** Booting Zephyr OS v1.12.0 *****
Initializing...
Bluetooth initialized
Mesh initialized
adv stack (real size 512):      unused 232      usage 280 / 512 (54 %)
ecc stack (real size 1024):     unused 104      usage 920 / 1024 (89 %)
prio recv thread stack (real size 448): unused 328      usage 120 / 448 (26 %)
recv thread stack (real size 1100):     unused 912      usage 188 / 1100 (17 %)`

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain:Zephyr SDK
 - nRF Mesh APP version: IOS V 1.0.4

jhedberg, 2019-05-06:
IIRC this app only has PB-ADV on the micro:bit (due to only 16k of RAM), whereas your provisioner app only supports PB-GATT. So that's the likely reason why it's not being discovered.

jhedberg, 2019-05-06:
One other note: there's a microbit_gatt.conf for this sample, however IIRC someone said the system footprint has grown too large to make it fit on the device. You might be able to fine-tune it though to make it fit.

closed 2019-05-06 by jhedberg