zephyrproject-rtos/zephyr #15924: Bluetooth: PTS: GATT server tests fail after merge of #15524

jhedberg, 2019-05-06:
From latest auto-pts run:
```
Regressions

GATT/SR/GAR/BV-04-C - Read Long Characteristic Value - from Server
GATT/SR/GAR/BV-06-C - Read Characteristic Descriptors - from Server
GATT/SR/GAR/BV-07-C - Read Long Characteristic Descriptor - from Server
GATT/SR/GAR/BV-08-C - Read Behind Long Characteristic Descriptor - from Server
GATT/SR/GAS/BV-01-C - Service Changed Characteristic - from Server
GATT/SR/GPA/BV-02-C - Secondary Service Declaration - from Server
GATT/SR/GPA/BV-03-C - Include Declaration - from Server
GATT/SR/GPA/BV-04-C - Characteristic Declaration - from Server
GATT/SR/GPA/BV-05-C - Characteristic Extended Property Declaration - from Server
GATT/SR/GPA/BV-06-C - Characteristic User Description Descriptor - from Server
GATT/SR/GPA/BV-07-C - Client Characteristic Configuration Desciptor - from Server
GATT/SR/GPA/BV-08-C - Server Characteristic Configuration Desciptor - from Server
GATT/SR/GPA/BV-12-C - Characteristic Format Descriptors - from Server
```

These are most likely caused by #15524 

@Vudentz suspects this is due to changed attribute handle values. Don't we communicate those using BTP so that them changing doesn't break things?

tedd-an, 2019-05-06:
I used git-bisect to narrow down and e0f3ab6b is the one that introduced the issue.

jhedberg, 2019-05-06:
> I used git-bisect to narrow down and e0f3ab6 is the one that introduced the issue.

@tedd-an yes, that's the same as #15524 that I've suspected from the beginning. Thanks for confirming that we're focusing on the right thing!

PR #15939 (2019-05-07): Bluetooth: gatt: Fix foreach iteration of static attributes
Fix calling bt_gatt_foreach_attr with start handle parameter set
to last static attribute handle.

Fixes #15924
closed 2019-05-07 by jhedberg