zephyrproject-rtos/zephyr #15941: Stale 1.3.99 documentation under /latest

jimparis, 2019-05-06:
There seem to be a bunch of stale documentation pages that report their version as "1.3.99" under /latest -- for example:

https://docs.zephyrproject.org/latest/api/io_interfaces.html
https://docs.zephyrproject.org/latest/api/device.html
https://docs.zephyrproject.org/latest/api/usb_api.html

I come across them occasionally through my own bookmarks and search engine results.  I would expect to at least get a 404 for these, although for things that have moved (by https://github.com/zephyrproject-rtos/zephyr/pull/12806?) a redirect to the new locations would be even more helpful.

jimparis, 2019-05-06:
Not just stuff under /latest/api either: e.g. https://docs.zephyrproject.org/latest/getting_started/getting_started.html

dbkinder, 2019-05-06:
Looks like the process to take the output of a doc build and copy to the AWS server is not cleaning out existing content (aren't we using rsync to the /latest folder)?

dbkinder, 2019-05-06:
... and looks like the detritus has been cleaned out now... (thanks @nashif )

carlescufi, 2019-05-08:
Fixed

closed 2019-05-08 by carlescufi