zephyrproject-rtos/zephyr #15968: rom_report very imprecise

oyvindronningstad, 2019-05-07:
**Describe the bug**
ninja rom_report doesn't give the correct number. When building hello_world on the board nrf52840_pca10056, rom_report gives a total (at the bottom) of 13236, but the symbols printed above only add up to ~11000.

**To Reproduce**
Steps to reproduce the behavior, either:
1. In samples/hello_world: `mkdir build; cd build`
2. `cmake -DBOARD=nrf52840_pca10056 -GNinja ..`
3. `ninja rom_report | grep -P "(^\S)" | grep -Po " \d+ " | paste -s -d+ - | bc`

This will print the sum of all the sizes of the top-level entries.

or:

1. In samples/hello_world: `mkdir build; cd build`
2. `cmake -DBOARD=nrf52840_pca10056 -GNinja ..`
3. Add `print(flash_symbols_total)` to the end of the `generate_target_memory_section()` function in the file scripts/footprint/size_report
4. `ninja rom_report`

This will print the internally calculated sum of the sizes of the symbols before the regular printout (scroll to the top).

In my case, it prints 10813.

**Expected behavior**
It's confusing that the symbols don't add up to the stated total at the bottom (in my case 13236).

**Impact**
It's imprecise enough that it's hard to do any precise footprint optimizations, i.e. it makes the feature close to useless.

**Environment (please complete the following information):**
 - OS: Ubuntu 18.10
 - Toolchain: Zephyr SDK
 - Commit SHA: 859c4ed2cd41aa0ef5b5a29b3d2c64199b252ee1

andrewboie, 2019-05-07:
Is the 'missing' space gaps between various symbols due to alignment?

oyvindronningstad, 2019-05-10:
> Is the 'missing' space gaps between various symbols due to alignment?

From looking into it, I think that is partly true, but I think there are certain symbols that somehow are not included at all. I think the size of a symbol should be reported with alignment filling, or the filling should also be reported so that the numbers match.

nashif, 2019-06-19:
@jocelyn-li  please assign

nashif, 2020-07-15:
@dcpleung  since you are at it, maybe this is already fixed?

nashif, 2020-08-27:
issues addressed in #26850

closed 2020-08-27 by nashif