zephyrproject-rtos/zephyr #15970: samples: microbit pong demo

maxxlife, 2019-05-07:
I'm trying to run demo for the micro:bit board called "Pong" game.
Unfortunately two boards can't work together.
Using minicom I discovered that one board says `Advertising successfully started                                                
Connected `
But game doesn't want to start.
I uploaded two .hex files to the two micro:bit boards. In prf.conf file I changed names of the boards to make a different name for each board.  My boards are connected via microUSB cables to have power supply.
Using buttons A and B I enter in "Connecting" mode on the screen of the micro:bit boards, but then they stuck in that mode.

**To Reproduce**
1. Build demo "pong" for the micro:bit board using west.
`source zephyr-env.sh`
`export ZEPHYR_TOOLCHAIN_VARIANT=zephyr`
`export ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk`
Go to the directory samples/boards/bbc_microbit/pong
and then run with the connected micro:bit to the computer
`west build -b bbc_microbit`
`west flash`
2. Using A button set mode "Multi" and then press B button to set mode "Connecting"
3. in prj.conf file change name of the board
4. Make steps 1 and 2 for the 2nd micro:bit board
5. Finally no game Pong. On the screen of the boards is just "Connecting" text. :(

**Expected behavior**
I expect that after boards are in "Connecting" mode, they should discover each other and connect, game Pong shoudl started.

**Screenshots or console output**
[image]

**Environment:**
 - OS: Linux Ubuntu 18.04
 - Toolchain Zephyr SDK 10

jhedberg, 2019-05-07:
Seems likely that this might be due to you not having #15939 in your tree

jhedberg, 2019-05-07:
@maxxlife confirmed (on slack) that this was indeed fixed by #15939

closed 2019-05-07 by jhedberg