zephyrproject-rtos/zephyr #15971: Fail to connect sample bluetooth HID with Tizen OS (BT_HCI_ERR_DIFF_TRANS_COLLISION)

Olivier-ProGlove, 2019-05-07:
I am trying to connect Nordic nRF52840 dev kit using last version of Zephyr (`859c4ed` west: Fix unused import and variable - Tue May 7) with [Samsung Gear S3](https://en.wikipedia.org/wiki/Samsung_Gear_S3) latest version of [Tizen OS](https://en.wikipedia.org/wiki/Tizen) 4.0.0.2.

**To Reproduce**
1. Build sample HID `samples/bluetooth/peripheral_hids` and flash on `nrf52840_pca10056`
2. Connect Tizen to Nordic DevKit
3. Tizen cannot connect and Zephyr returns the error `<inf> main: Disconnected from  (reason 42)`

**Expected behavior**
It is expected Tizen OS to be able to connect to Zephyr board.
FYI, I removed passphrase in the sample HID. So no passphrase is expected to be asked and I confirm it works with other OSes.

**Impact**
Zephyr Bluetooth devices cannot be connected to Tizen OS.

**Screenshots or console output**
I have duplicated the issue two times. I have added the `btmon` traces and the Wireshark Nordic's sniffer traces (I cannot see the disconnection packet in the sniffer trace): 
[tizen-github-issue1.zip](https://github.com/zephyrproject-rtos/zephyr/files/3152812/tizen-github-issue1.zip)

The log (with the default logging level) does not show much:

```
***** Booting Zephyr OS zephyr-v1.14.0-578-g7abd268e9db2 *****
Bluetooth initialized
Advertising successfully started
[00:00:00.199,300] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.199,300] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.199,300] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 1.14 Build 99
[00:00:00.247,600] <wrn> bt_hci_core: No ID address. App must call settings_load()
[00:00:00.283,000] <inf> bt_hci_core: Identity: ea:ac:67:83:44:59 (random)
[00:00:00.283,000] <inf> bt_hci_core: HCI: version 5.0 (0x09) revision 0x0000, manufacturer 0x05f1
[00:00:00.283,000] <inf> bt_hci_core: LMP: version 5.0 (0x09) subver 0xffff
[00:01:13.410,700] <inf> main: Connected to 'e4:d8:e3:3a:14:5a (random)'
[00:01:13.673,000] <inf> main: Disconnected from  (reason 42)
```

**Environment (please complete the following information):**
 - OS: Tizen 4.0.0.2 (I was told it was working with Tizen 3.x)
 - Latest Zephyr: `859c4ed` west: Fix unused import and variable - Tue May 7

cc:  @carlescufi @joerchan @cvinayak

joerchan, 2019-05-07:
@Olivier-ProGlove This seems to be a bug in the phone side.
It initiates 2 LL procedures with overlapping instant. The Zephyr controller is correct in dropping the link with error code transaction collision.

Procedures in error (sniffer-trace1)
3: PHY_UPDATE_IND(instant = 9)
5: CONN_UPDATE_REQ(instant = 12)

Olivier-ProGlove, 2019-05-07:
It looks you are right. I tried to disable `CONFIG_BT_PHY_UPDATE=n` and I could connect to Tizen OS.

I actually noticed after testing this workaround a similar bug existed for Samsung S8: https://github.com/zephyrproject-rtos/zephyr/issues/13396 with the same workaround :-)

closed 2019-05-07 by Olivier-ProGlove