zephyrproject-rtos/zephyr #15975: Openthread - fault with dual network interfaces

acsplayon, 2019-05-07:
**Describe the bug**
Multiple network interfaces (ETHERNET and OPENTHREAD) cause hard fault in openthread network_management callback.

**To Reproduce**
Steps to reproduce the behavior:
1) You create 2 L2 network interfaces. ETHERNET and OPENTHREAD
2) I use USB cdc_ecm to interface to the ETHERNET iface.
3) no special configuration of openthread.
4) on init, openthread will install a net_mgmt callback which is located in openthread.c.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Showstopper. System freezes/is unstable

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: Linux (ubuntu 16.04/18.04)
 - Toolchain Zephyr SDK
 - Commit - master

**Additional context**
ipv6_addr_event_hander at/near openthread.c:67 will be called by net management events for all net ifaces.  This event handler does no checking on whether the context returned by net_if_l2_data(iface) is a valid openthread_context and a hard fault will be issued eventually in a memcmp call.

I added a u32_t to the openthread context called ot_magic, and initialized it in openthread_init to hold a magic number/identifier.  Then, in ipv6_addr_even_handler, I check the returned data structure for both NULL and the magic number, and if either fail, skip the processing of the IPV6 address on the invalid openthread context.

jukkar, 2019-05-09:
> I added a u32_t to the openthread context called ot_magic, and initialized it in openthread_init to hold a magic number/identifier. Then, in ipv6_addr_even_handler, I check the returned data [...]

Could you send a PR if you have already a fix?

rlubos, 2019-07-03:
@acsplayon I've reproduced the issue and submitted https://github.com/zephyrproject-rtos/zephyr/pull/17308 to fix it. Please review/verify.

PR #17308 (2019-07-03): net: openthread: Verify iface in net_mgmt event handler
OpenThread did not verify if the interface provided in the net_mgmt
handler is actually an OpenThread interface. In result, when multiple
network interfaces were used, different interfaces were processed by the
OpenThread handler, ending up in a crash.

Fixes #15975.
PR #17315 (2019-08-15): [Backport v1.14-branch] net: openthread: Verify iface in net_mgmt event handler
Backport #17308.

Fixes #15975.
closed 2019-07-03 by jukkar