zephyrproject-rtos/zephyr #15985: [Coverity CID :198395]Memory - corruptions in /soc/arm/microchip_mec/mec1501/soc.c

aasthagr, 2019-05-07:
Static code scan issues seen in File: /soc/arm/microchip_mec/mec1501/soc.c
 Category: Memory - corruptions
 Function: soc_ecia_init
 Component: Other
 CID: 198395
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-05-08:
```
110     	/* gate on GIRQ's that are aggregated only */
111     	ECIA_REGS->BLK_EN_SET = MCHP_ECIA_AGGR_BITMAP;
112     
113     	/* Clear all GIRQn source enables and source status */
114     	pg = &ECIA_REGS->GIRQ08;
115     	for (n = 0u; n < MCHP_GIRQ_ZID_MAX; n++) {
>>>     CID 198395:  Memory - corruptions  (OVERRUN)
>>>     Overrunning array of 1 20-byte elements at element index 18 (byte offset 360) by dereferencing pointer "pg".
116     		pg->EN_CLR = 0xFFFFFFFFul;
117     		pg->SRC = 0xFFFFFFFFul;
118     		pg++;
119     	}
120     
121     	/* Clear all external NVIC enables and pending status */

```

ioannisg, 2019-06-30:
@andrewboie have you looked at this one?

andrewboie, 2019-07-02:
@franciscomunoz @albertofloyd can you take a look at this one?

ioannisg, 2019-07-27:
This is reported as medium impact bug, so please, take a look if possible (seems to be an easy fix)

ioannisg, 2019-07-31:
@franciscomunoz @albertofloyd ping

albertofloyd, 2019-08-01:
Will submit fix tomorrow

scottwcpg, 2019-08-01:
Pushed PR 17950 to fix.

PR #17950 (2019-08-01): soc : arm : microchip_mec Fix MEC1501 SoC ECIA initialization
Changes to MEC1501 HAL for eSPI changed loop constant used by SoC
code initializing the ECIA registers. Fixed by using HAL constants
that are invariant.

Fixes #15985
closed 2019-08-01 by ioannisg