zephyrproject-rtos/zephyr #15986: [Coverity CID :198394]Error handling issues in /tests/net/socket/socket_helpers.h

aasthagr, 2019-05-07:
Static code scan issues seen in File: /tests/net/socket/socket_helpers.h
 Category: Error handling issues
 Function: prepare_listen_sock_udp_v4
 Component: Tests
 CID: 198394
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-05-08:
```
/tests/net/socket/socket_helpers.h: 26 in prepare_listen_sock_udp_v4()
20     	zassert_true(ret >= 0, "socket open failed");
21     
22     	sock = ret;
23     
24     	zassert_equal(addr->sin_family, AF_INET, "Invalid family");
25     
>>>     CID 198394:  Error handling issues  (NEGATIVE_RETURNS)
>>>     "sock" is passed to a parameter that cannot be negative. [Note: 
>>> The source code implementation of the function has been overridden 
>>> by a builtin model.]
26     	ret = bind(sock, (struct sockaddr *)addr, sizeof(*addr));
27     	zassert_equal(ret, 0, "bind failed (%d/%d)", ret, errno);
28     
29     	return sock;
30     }
31     

```

jukkar, 2019-05-12:
False positive

closed 2019-05-12 by jukkar