zephyrproject-rtos/zephyr #15987: [Coverity CID :198393]Error handling issues in /tests/net/socket/socket_helpers.h

aasthagr, 2019-05-07:
Static code scan issues seen in File: /tests/net/socket/socket_helpers.h
 Category: Error handling issues
 Function: prepare_listen_sock_udp_v6
 Component: Tests
 CID: 198393
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-05-08:
```
/tests/net/socket/socket_helpers.h: 45 in prepare_listen_sock_udp_v6()
39     	zassert_true(ret >= 0, "socket open failed");
40     
41     	sock = ret;
42     
43     	zassert_equal(addr->sin6_family, AF_INET6, "Invalid family");
44     
>>>     CID 198393:  Error handling issues  (NEGATIVE_RETURNS)
>>>     "sock" is passed to a parameter that cannot be negative. [Note: 
>>> The source code implementation of the function has been overridden 
>>> by a builtin model.]
45     	ret = bind(sock, (struct sockaddr *)addr, sizeof(*addr));
46     	zassert_equal(ret, 0, "bind failed (%d/%d)", ret, errno);
47     
48     	return sock;
49     }
50     
```

jukkar, 2019-05-12:
False positive

closed 2019-05-12 by jukkar