zephyrproject-rtos/zephyr #15988: [Coverity CID :198392]Insecure data handling in /tests/net/socket/getaddrinfo/src/main.c

aasthagr, 2019-05-07:
Static code scan issues seen in File: /tests/net/socket/getaddrinfo/src/main.c
 Category: Insecure data handling
 Function: process_dns
 Component: Tests
 CID: 198392
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

nashif, 2019-05-08:
```
/tests/net/socket/getaddrinfo/src/main.c: 151 in process_dns()
145     						errno);
146     					break;
147     				}
148     
149     				NET_DBG("Received DNS query");
150     
>>>     CID 198392:  Insecure data handling  (TAINTED_SCALAR)
>>>     Passing tainted variable "recv_buf" to a tainted sink.
151     				ret = check_dns_query(recv_buf,
152     						      sizeof(recv_buf));
153     				if (ret) {
154     					sys_mutex_unlock(&wait_data);
155     				}
156     	
```

jukkar, 2019-05-12:
Tainting the variable is not really an issue in this test application because we know what we are receiving. There is nothing here to fix.

closed 2019-05-12 by jukkar