zephyrproject-rtos/zephyr #16000: We need a CI check for commas in CODEOWNERS

aescolar, 2019-05-08:
We need a new check in CI to prevent commas being added to the CODEOWNERS files.

Background:
CODEOWNERS has been broken twice (#15998 , #15709) in the last weeks.
The combination of DTS files having commas in their names, and the new checks for missing owners to files, means authors have and will continue trying to add lines to CODEOWNERS for particular files which include commas.
This breaks CODEOWNERS, as GitHub cannot handle files with commas.

Other options considered:
* Educating authors to not add commas: Unlikely to be reliable as the new setup encourages any author to add himself to CODEOWNERS, including very non engaged ones
* Reviewers checking for it: Tedious and therefore error prone.
* Reacting/fixing it when it breaks: Tedious

marc-hb, 2019-05-08:
I wonder which unrelated syntax commas were inspired by. Shell's brace expansion maybe?

Checking for commas is useful but it won't catch other sorts of errors. When I implemented the latest version of the CI check I looked for a way to re-use git ignore and/or git attributes but didn't find any "API".

In the longer term what all github users (not just Zephyr) need is github to make the CODEOWNERS more usable:
- report errors
- not (silently) ignore the entire file when just one line is wrong
- the best and final: share their parser.

Is there an issue tracker for github itself? I couldn't find one. The github blog introducing CODEOWNERS points at this support form: https://github.com/contact?form%5Bsubject%5D=Code%20Owners

Someone already did contact support but that didn't seem to go far:
https://lists.zephyrproject.org/g/devel/message/5950
Maybe he wasn't a paying customer and didn't have access to the enterprise/premium support?

marc-hb, 2019-05-08:
> I wonder which unrelated syntax commas were inspired by. Shell's brace expansion maybe?

Doh, just realized some filenames have actual commas in them. Sorry.

So there's another github problem far worse than silently failing to parse invalid CODEOWNERS: Github is actually failing to parse valid CODEOWNERS. This should really be reported to github.

> Reviewers checking for it: Tedious and therefore error prone.

What does github do when a single commit hits multiple CODEOWNERS lines? Does it automatically request reviews from the owners on all the lines? If yes then it's worth adding a CODEOWNERS line inside the CODEOWNERS file itself.

PS: still not sure why CODEOWNERS has to be fine-grained down to the file level, feels like "extreme decentralization". I expected most of CODEOWNERS to be directories - and the file to be shorter. Off topic sorry.

aescolar, 2019-05-08:
@marc-hb In the long term. But in the short term, a stopgap like:
```KK=`grep -v '^#'  CODEOWNERS | grep ","`; if [ ! -z "$KK" ]; then echo "CODEOWNERS file cannot have entries with commas" ; exit 1; fi; ```
in the relevant CI script could save us some pain

marc-hb, 2019-05-20:
It should indeed be easy (famous last words) to implement something like this in `check_compliance.py`, feel free to submit a PR.

However I'd like the long term to at least *start* before ~sinking~ spending time in the "whack-a-mole" short term. Having some support ticket filed could/would:
- hopefully stop the long term from slipping every day;
- get github's advice for a better short term, for instance a list of other failing characters;
- provide a centralized, non-Zephyr place to discuss (CODEOWNERS is not specific and unrelated to Zephyr);
- provide a non-Zephyr, long term ticket number that can be referenced in every short term workaround.

Maybe someone in the TSC would know someone who can file this support request? :-)

> I expected most of CODEOWNERS to be directories

Maybe not that off-topic after all. Seems like better CODEOWNERS practice and would avoid this issue entirely.

marc-hb, 2019-07-06:
@nashif just added this check in https://github.com/zephyrproject-rtos/ci-tools/commit/700265513296779884

closed 2019-07-07 by aescolar