zephyrproject-rtos/zephyr #16023: dfu: enabling usb dfu in MCUboot crashes zephyr application in slot0

jli157, 2019-05-08:
**Describe the bug**
A clear and concise description of what the bug is.
With USB cable always connected to a computer, the zephyr application, regardless which one, always crashes at booting up if USB DFU is enabled in MCUBOOT. 

**To Reproduce**
Steps to reproduce the behavior:
1. apply PR #15949 so that nucleo_f429zi has a partition table.
2. Enable DFU feature in mcuboot by creating a file "$MCUBOOT/boot/zephyr/boards/nucleo_f429zi.conf" with the following contents:
```
# Required
CONFIG_MULTITHREADING=y
CONFIG_BUILD_OUTPUT_HEX=y

#Enable DFU
CONFIG_BOOT_WAIT_FOR_USB_DFU=y

# USB
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="MCUBOOT DFU"
CONFIG_USB_DFU_CLASS=y
CONFIG_FLASH_PAGE_LAYOUT=y
```

2. build mcuboot and flash it to a nucleo_f429zi board. 
3. cd samples/subsys/usb/dfu
4. west build -b nucleo_f429zi
5. generate a signed firmare: 
     `west sign -t imgtool -- --key $MCUBOOT/root-rsa-2048.pem`
6. flash the signed hex file `zephyr.signed.hex`. So far `west flash --hex-file` doesn't work for this purpose. I ended up using [`st-flash`](https://github.com/texane/stlink) to flash the signed hex file. 
7. Make sure the system USB is **always connected** to the board. 
8. reboot the system
9. You'll see the error messages. 

I tried different zephyr applications with 
**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
DFU can't be enabled in MCUBoot. However, DFU is useful for populating firmware through MCUboot. 

**Screenshots or console output**
```
***** Booting Zephyr OS zephyr-v1.14.0-655-g893c4f5d7363 *****
[00:00:00.064,000] <inf> mcuboot: Starting bootloader
[00:00:00.069,000] <inf> mcuboot: Primary image: magic=unset, copy_done=0x3, image_ok=0x3
[00:00:00.078,000] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.086,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.096,000] <inf> mcuboot: Swap type: none
[00:00:00.166,000] <inf> mcuboot: Bootloader chainload address offset: 0x20000
[00:00:00.173,000] <inf> mcuboot: Jumping to the first image slot
***** Booting Zephyr OS zephyr-v1.14.0-656-g6bcb8d5538ca *****
***** BUS FAULT *****
  Precise data bus error
  BFAR Address: 0xef7fbc2
***** Hardware exception *****
Current thread ID = 0x20000a6c
Faulting instruction address = 0x8027f6e
Fatal fault in ISR! Spinning...
```
**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain Zephyr-sdk-0.10.0
 - Commit SHA or Version used on zephyr: 95e1eb9
- Commit SHA or Version used on MCUBoot: e1727d9d2bf53e370b10b9cc8761a96e70a983ab

jli157, 2019-05-08:
zephyr.lst is attached for analyzing. 

[zephyr_crashed_by_mcuboot.lst.txt](https://github.com/zephyrproject-rtos/zephyr/files/3159352/zephyr_crashed_by_mcuboot.lst.txt)

jli157, 2019-05-08:
@jfischer-phytec-iot can you help to check this one? This is really a blocking issue for me.

jfischer-phytec-iot, 2019-05-08:
@jli157 Can you please first verify that it works to boot other sample e.g. hello_world on your platform?

jli157, 2019-05-09:
> @jli157 Can you please first verify that it works to boot other sample e.g. hello_world on your platform?

I got the same issue with `hello_world` as well, please check the logs below:

```
***** Booting Zephyr OS zephyr-v1.14.0-655-g893c4f5d7363 *****
[00:00:00.064,000] <inf> mcuboot: Starting bootloader
[00:00:00.069,000] <inf> mcuboot: Primary image: magic=unset, copy_done=0x3, image_ok=0x3
[00:00:00.078,000] <inf> mcuboot: Scratch: magic=unset, copy_done=0x0, image_ok=0x3
[00:00:00.086,000] <inf> mcuboot: Boot source: primary slot
[00:00:00.096,000] <inf> mcuboot: Swap type: none
[00:00:00.149,000] <inf> mcuboot: Bootloader chainload address offset: 0x20000
[00:00:00.156,000] <inf> mcuboot: Jumping to the first image slot
***** Booting Zephyr OS zephyr-v1.14.0-656-g6bcb8d5538ca *****
Hello World! nucleo_f429zi
***** Spurious interrupt (IRQ  67) *****
***** Hardware exception *****
Current thread ID = 0x20000158
Faulting instruction address = 0x8021084
Fatal fault in ISR! Spinning...

```

jli157, 2019-05-09:
checked the address 0x8021084 in zephyr.lst for `hello_word`:

```
0802107c <k_cpu_idle>:

#if defined(CONFIG_ARMV6_M_ARMV8_M_BASELINE)
        cpsie i
#elif defined(CONFIG_ARMV7_M_ARMV8_M_MAINLINE)
        /* clear BASEPRI so wfi is awakened by incoming interrupts */
        eors.n r0, r0
 802107c:       4040            eors    r0, r0
        msr BASEPRI, r0
 802107e:       f380 8811       msr     BASEPRI, r0
#else
#error Unknown ARM architecture
#endif /* CONFIG_ARMV6_M_ARMV8_M_BASELINE */

        wfi
 8021082:       bf30            wfi

        bx lr
 8021084:       4770            bx      lr
 8021086:       bf00            nop

```

jli157, 2019-05-09:
I used "CONFIG_USB_CDC_ACM=y" instead of "CONFIG_USB_DFU_CLASS=y", and still got the same issue. Maybe it is a USB device issue from ST Micro's stack?

jli157, 2019-05-09:
The spurious IRQ67 on STM32F4XX is the USB_FS interrupt. I guess the enabled USB interrupt disturbed the booting process. 

Ref: [page 373 on RM](https://www.st.com/content/ccc/resource/technical/document/reference_manual/3d/6d/5a/66/b4/99/40/d4/DM00031020.pdf/files/DM00031020.pdf/jcr:content/translations/en.DM00031020.pdf#page371)

67  OTG_FS USB On The Go FS global interrupt

lemrey, 2019-05-16:
Hi @jli157, if you are trying to transfer an image using MCUboot's Serial Recovery mode over USB, you'll need to turn on `CONFIG_BOOT_SERIAL_CDC_ACM` (which in turn selects `CONFIG_USB_CDC_ACM`). Note, this does not use the USB DFU class.

Then, you must be able to disable USB from firing interrupts (`usb_disable`) otherwise those, if not handled by the application, will crash.

If you want to use the USB DFU class to transfer a firmware image, then I don't think you'll need to turn on MCUboot's Serial Recovery mode, or USB support, at all.

jli157, 2019-05-16:
> Hi @jli157, if you are trying to transfer an image using MCUboot's Serial Recovery mode over USB, you'll need to turn on `CONFIG_BOOT_SERIAL_CDC_ACM` (which in turn selects `CONFIG_USB_CDC_ACM`). [...]

Do you mean CONFIG_MCUBOOT_SERIAL as the recovery mode? Actually, I'm not using either the recovery mode or CONFIG_BOOT_SERIAL_CDC_ACM. I'm using the DFU mode by enabling the option `CONFIG_BOOT_WAIT_FOR_USB_DFU` which will directly uses the DFU module implemented by Zephyr (subsys/usb/dfu) to transfer a firmware image to the slot1. I think it is reasonable to use the DFU mode as well as  the `MCUBOOT_SERIAL` since both are just transferring an updating image to the slot1, right?  So, to use the DFU mode, usb functionality should be disabled before leaving MCUBoot to jump to Zephyr application. This is why this issue happened.

erwango, 2019-11-01:
@jli157, is the issue still active?

d3zd3z, 2019-12-03:
Is [MCUboot pr#478](https://github.com/JuulLabs-OSS/mcuboot/pull/478) sufficient to fix this? If not, this should probably be filed as an issue with MCUboot, as it should not be leaving the hardware in a state where unexpected interrupts can come in.

erwango, 2020-02-12:
> Is MCUboot pr#478 sufficient to fix this? If not, this should probably be filed as an issue with MCUboot, as it should not be leaving the hardware in a state where unexpected interrupts can come in.

@jli157 can you  check ?

nvlsianpu, 2020-02-13:
I found that fixed since a while. Just tested again with success (on nrf52840_pca10056).

closed 2020-02-13 by nvlsianpu