zephyrproject-rtos/zephyr #16031: Toolchain abstraction

mped-oticon, 2019-05-09:
Zephyr currently assumes standard open source toolchains, such a GNU or LLVM.

There is a desire to have Zephyr's CMake build system support more toolchains - including proprietary commercial ones. 
Furthermore, the name of some of these proprietary commercial toolchains cannot be due to industry competition. Hence out-of-tree support for any toolchain is also required.

Rather than polluting CMake code with branching out for every supported toolchain, abstraction interfaces will be made.

Solution proposals and further motivation have been presented to the TSC:
https://drive.google.com/file/d/1hGgk9ITFTXS3QoOr3GTFlC46VhEopT0Y/view?usp=sharing 

The aim is to abstract any hardcoded references to {compiler, linker, binutils} out.
We want to retain as much of the CMake flow as possible, to avoid toolchains drifting too far apart.

Related discussion: https://github.com/zephyrproject-rtos/zephyr/issues/10182

mped-oticon, 2019-05-09:
Toolchain abstraction has been spanning quite a few PRs:
https://github.com/zephyrproject-rtos/zephyr/pulls?utf8=%E2%9C%93&q=%22Toolchain+abstraction%22

aescolar, 2019-06-26:
This is still work in progress

aescolar, 2019-07-11:
Moved milestone to 2.1 as it is quite clear not all pending work will be completed by 2.0

MaureenHelm, 2020-06-09:
@tejlmand can you please update this issue?

tejlmand, 2020-06-30:
@MaureenHelm A draft PR has been made: https://github.com/zephyrproject-rtos/zephyr/pull/24851

Based on feedback in the PR and discussions in the Toolchain WG, this appears to be the right way forward.

tejlmand, 2020-09-24:
New abstraction is in place.

closed 2020-09-24 by tejlmand